bmax121/APatch #629: Better Internet Tiles - superuser for module disappears after reboot

burntoc, 2024-08-25:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Just the title. I've been running on Apatch on my P7P with latest updates for about a year with no issues, a half dozen or so modules all good.  Only one, Better Internet Tiles, loses superuser every time I reboot.  

### Reproduce method/复现方法

1. APatch -> Superuser -> Grant su to module
2. Reboot
3. Check su, it will be disabled.

### Expected behavior/预期行为

Expected to persist through reboots.

### Actual behaviour /实际行为

Superuser rights disappear.

### Device Name/设备名称

Pixel 7 Pro

### OS Version/系统版本

14, latest

### APatch Version/APatch 版本

latest

### Kernel Version/内核版本

5.10.198

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

Tried, failed so far.

xhxjsk, 2024-08-27:
Notes
The newly designed exclusion module modifies the logic, and from now on, applications that have already been granted ROOT permissions no longer support exclusion because it does not conform to the logic of normal usage scenarios; By default, modules will no longer be excluded because not all software will check the current environment, which undoubtedly brings unnecessary power consumption and heat to the phone. You can manually configure the applications that need to be excluded. It should be noted that if you are upgrading from a version of KernelPatch below 0.10.7 instead of a completely new installation, you need to manually delete the /data/adb/ap/package_config file and reconfigure it on the SuperUser screen, otherwise a BUG may be occurred.
We have noticed that some older devices (mainly those with Linux kernel versions less than 5) may experience unlimited loading of SuperUser Screen, and even some applications may not be able to use ROOT permissions. This is a known issue and is currently under investigation.

burntoc, 2024-08-29:
> Notes The newly designed exclusion module modifies the logic, and from now on, applications that have already been granted ROOT permissions no longer support exclusion because it does not conform to the logic of normal usage scenarios; ...

I might be misunderstanding the response, but I'm not trying to exclude anything.  None of my apps are granted root by default with Apatch - I must select them.  Of 140+ apps I have about 10 I've granted superuser and this persists across shutdowns, reboots, etc.  BIT is the only module that changes to disabled when the machine is shutdown or rebooted.

closed 2025-03-09 by Admirepowered