bmax121/APatch #631: 10918每次重启应用掉root权限，超级用户授权列表恢复默认

wzgrx, 2024-08-26:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

10918每次重启应用掉root权限，超级用户授权列表恢复默认，已经授权的应用，会取消授权

### Reproduce method/复现方法

重启

### Expected behavior/预期行为

正常

### Actual behaviour /实际行为

掉root

### Device Name/设备名称

note12 t

### OS Version/系统版本

14

### APatch Version/APatch 版本

10918

### Kernel Version/内核版本

5.10

### KernelPatch Version/KernelPatch 版本

0.11.1

### Additional context/其他信息

0kb

Ai686Leo, 2024-09-02:
一加11也有此现象，目前受影响应用就一个
[Uploading APatch_bugreport_2024-09-02_22_16.tar.gz…]()

pomelohan, 2024-09-12:
提供 APatch bugreport