bmax121/APatch #646: Random/Editable package name & app name

skhrlx, 2024-09-01:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

Not directly, but is a question of time to some ppl start looking for APatch and triggering an detection

### Describe the solution you'd like/描述你想要的解决方案

An solution like Magisk give us, the possibility to change app name and package name.

### Describe alternatives you've considered/描述您考虑过的备选方案

At least the app name change hability

crypticc1, 2024-09-01:
Same as this one? 

https://github.com/bmax121/APatch/issues/544

skhrlx, 2024-09-01:
> Same as this one?
> 
> #544

Oh, yes, (sorry for the duplicate devs), but it's something that we need fastly since the growth of APatch

1001ace1001, 2024-09-02:
just uninstall manager.

skhrlx, 2024-09-03:
> just uninstall manager.

Wow! Incredible solution from "1001ace1001" 🤓☝️

marat2509, 2024-12-14:
https://github.com/bmax121/APatch/issues/544

closed 2024-12-14 by marat2509