bmax121/APatch #649: Momo : the environment is abnormal

abda11ah, 2024-09-03:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

I'm using the latest nightly version of Apatch 
When I enable "enforce denylist" or install zygisk assistant, momo tells me that the environment is broken
[image]
.

### Reproduce method/复现方法

Install apatch nightly 10916
enable enforce denylist
Exclude modifications to momo app.

### Expected behavior/预期行为

Momo tells me Magisk or Zygisk are not installed

### Actual behaviour /实际行为

When i uncheck exclude modifications, momo tells me Magisk is detected
[image]

### Device Name/设备名称

Xiaomi mi 10T lite 5G (Gauguin)

### OS Version/系统版本

14

### APatch Version/APatch 版本

10916

### Kernel Version/内核版本

4.19.300 perf

### KernelPatch Version/KernelPatch 版本

10916

### Additional context/其他信息

Magisk assistant give the same behavior as "enforce denylist" option

doandat943, 2024-11-01:
I have same problem with you

zalnaRs, 2024-11-02:
You need to hide root. There is some module which you need to install. Chiteroman showcased it on his Play Integrity Fix telegram channel

yoshimo, 2024-11-02:
people assume that using a kernel patch and a super key will not require additional tools to hide root.

doandat943, 2024-11-02:
> You need to hide root. There is some module which you need to install. Chiteroman showcased it on his Play Integrity Fix telegram channel

i have installed it, but no affect

closed 2025-03-09 by Admirepowered