bmax121/APatch #651: APatch最新测试版10923已嵌入内核模块却不显示

Fyg369, 2024-09-05:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

内核模块已嵌入不显示
[image]
[image]

### Reproduce method/复现方法

卸载旧版-安装APatch最新测试版10923-开始boot打补丁和嵌入内核模块-刷入打好补丁和内核模块boot

### Expected behavior/预期行为

[image]
[image]
[image]

### Actual behaviour /实际行为

已经嵌入内核模块却不显示模块

### Screenshots/截图

[image]
[image]

### Device Name/设备名称

一加6T

### OS Version/系统版本

安卓11

### APatch Version/APatch 版本

测试版10923

### Kernel Version/内核版本

4.9.250

### KernelPatch Version/KernelPatch 版本

0.11.1

### Additional context/其他信息

[image]

pomelohan, 2024-09-11:
并非 bug，嵌入是没有问题的，只是这个 KPM 不适合你的设备，开机的时候无法加载，自然就不在模块列表中，但仍然在启动镜像中

closed 2024-09-11 by pomelohan

Fyg369, 2024-09-11:
> 不是bug，嵌入是没有问题的，只是这个KPM不适合你的设备，开机的时候无法加载，自然不会在模块列表中，但仍然在启动镜像中

[image]
[image]