bmax121/APatch #662: option not to reboot after ota

philfry, 2024-09-11:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

I have a Oneplus 8T. When doing a system upgrade, after the Oneplus app has downloaded, extracted and verified the firmware,  I usually have to press "install now" for the firmware to be installed. The phone instantly reboots then and APatch is gone.
If, instead of clicking "install now", I enter the APatch app and install APatch into the "inactive slot (after ota)", I get a warning that the phone will be force-rebooted. Which is unfortunate, because then the firmware update would not be installed (I had the same behaviour with Magisk).

### Describe the solution you'd like/描述你想要的解决方案

Make "reboot after installation into the inactive slot (ota)" optional.

### Describe alternatives you've considered/描述您考虑过的备选方案

My workaround for now is to extract the `boot.img` from the firmware, patch it and install it via fastboot after the firmware update is done.

Guation, 2024-11-07:
K50一样的情况 修补之后重启系统不会升级 只能先把boot提取出来 升级完事后重新修补刷入

philfry, 2024-11-07:
I don't know whether or not I got you right, but in Magisk it worked this way:

1. let the updater fetch the ota image
2. wait for extraction to complete
3. wait for verification to complete
4. ignore the updater's prompt "install now"
5. switch to Magisk and install Magisk into the "inactive slot (after OTA)"
6. ignore Magisk's "restart now" 
7. switch back to the updater and press "install now"
8. wait for the reboot to complete and enjoy an updated and rooted OS

With APatch I cannot switch back to the installer at step 7 because there's no prompt like in step 6 that I could ignore. APatch simply restarts the phone.

For me it would be sufficient to just add a prompt "reboot now" instead of force-rebooting the phone after installing APatch into the inactive slot. Would that be an option for you as well?

psy0ch, 2025-03-04:
still a issue, I had to manually repatch it last time and it seems to not have changed with the latest update

yezhiyi9670, 2025-09-17:
As far as I know, APatch does not automatically reboot after patching the kernel in the inactive slot. It only warns you that it will switch slot _when you reboot the device_, so the method you use with Magisk will still work.

Furthermore, I guess (**not 100% sure**) what the "install now" button in the OTA app does is just switching slot and rebooting. The OTA image is probably already written into the inactive slot during the "extracting" phase.