bmax121/APatch #666: Can't patch bootimage

Niemandausduisburg, 2024-09-13:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

I can't patch my bootimage.

09-13 18:21:58.622 19025 19025 D PatchViewModel: ****************************
09-13 18:21:58.638 19025 19025 D PatchViewModel: ****************************
09-13 18:21:58.638 19025 19025 D PatchViewModel:  APatch Boot Image Patcher
09-13 18:21:58.639 19025 19025 D PatchViewModel: ****************************
09-13 18:21:58.639 19025 19025 D PatchViewModel: - Patching kernel
09-13 18:21:58.639 19025 19025 D PatchViewModel: + ./kptools -p -i kernel.ori -S Sc4n33w177c43n -k kpimg -o kernel -K kpatch
09-13 18:21:58.897 19025 19025 D PatchViewModel: [-] /home/runner/work/KernelPatch/KernelPatch/tools/patch.c:367/patch_update_img(); analyze_kallsym_info error
09-13 18:21:58.897 19025 19025 D PatchViewModel: [+] kernel image_size: 0x026a2810
09-13 18:21:58.897 19025 19025 D PatchViewModel: [+] kernel uefi header: false
09-13 18:21:58.897 19025 19025 D PatchViewModel: [+] kernel load_offset: 0x00080000
09-13 18:21:58.897 19025 19025 D PatchViewModel: [+] kernel kernel_size: 0x031cf000
09-13 18:21:58.897 19025 19025 D PatchViewModel: [+] kernel page_shift: 12
09-13 18:21:58.897 19025 19025 D PatchViewModel: [+] new kernel image ...
09-13 18:21:58.897 19025 19025 D PatchViewModel: [+] linux_banner 1: Linux version 4.19.246-N0kernel (nobody@android-build) (Android (11368308, +pgo, +bolt, +lto, +mlgo, based on r510928) clang version 18.0.0 (https://android.googlesource.com/toolchain/llvm-project 477610d4d0d988e69dbc3fae4fe86bff3f07f2b5), LLD 18.0.0) #2 SMP PREEMPT Fri Sep 13 17:44:28 CEST 2024
09-13 18:21:58.897 19025 19025 D PatchViewModel: [+] linux_banner offset: 0x1d39adc
09-13 18:21:58.897 19025 19025 D PatchViewModel: [+] kernel version major: 4, minor: 19, patch: 246
09-13 18:21:58.897 19025 19025 D PatchViewModel: [+] kallsyms_token_table offset: 0x01b09e00
09-13 18:21:58.897 19025 19025 D PatchViewModel: [+] endian: little
09-13 18:21:58.897 19025 19025 D PatchViewModel: [+] kallsyms_token_index offset: 0x01b0a200
09-13 18:21:58.897 19025 19025 D PatchViewModel: [+] find arm64 relocation kernel_va: 0xffffff8010000000
09-13 18:21:58.897 19025 19025 D PatchViewModel: [+] find arm64 relocation table range: [0x02172f18, 0x02402238), count: 0x0001b4cc
09-13 18:21:58.897 19025 19025 D PatchViewModel: [?] bad rela offset: 0xffffff80126ce6d0
09-13 18:21:58.897 19025 19025 D PatchViewModel: [-] /home/runner/work/KernelPatch/KernelPatch/tools/kallsym.c:470/find_markers_1(); find kallsyms_markers error
09-13 18:21:58.897 19025 19025 D PatchViewModel: [+] kallsyms_markers range: [0x0196a600, 0x0196a960), count: 0x0000006b
09-13 18:21:58.898 19025 19025 D PatchViewModel: [+] approximate kallsyms_offsets range: [0x0196a5e0, 0x019c313c) count: 0x000162d7
09-13 18:21:58.898 19025 19025 D PatchViewModel: [-] /home/runner/work/KernelPatch/KernelPatch/tools/kallsym.c:583/find_names(); find kallsyms_names error
09-13 18:21:58.898 19025 19025 D PatchViewModel: [-] /home/runner/work/KernelPatch/KernelPatch/tools/kallsym.c:470/find_markers_1(); find kallsyms_markers error
09-13 18:21:58.898 19025 19025 D PatchViewModel: [+] kallsyms_markers range: [0x0196a600, 0x0196a960), count: 0x0000006b
09-13 18:21:58.898 19025 19025 D PatchViewModel: [+] approximate kallsyms_offsets range: [0x0196a5e0, 0x019c313c) count: 0x000162d7
09-13 18:21:58.898 19025 19025 D PatchViewModel: [-] /home/runner/work/KernelPatch/KernelPatch/tools/kallsym.c:583/find_names(); find kallsyms_names error
09-13 18:21:58.914 19025 19025 D PatchViewModel: + patch_rc=1
09-13 18:21:58.914 19025 19025 D PatchViewModel: + set +x
09-13 18:21:58.914 19025 19025 D PatchViewModel: - Patch kernel error: 1
09-13 18:21:58.933  4414 24256 W MediaProvider: isAppCloneUserPair failed. Users: 0 and 0
09-13 18:21:58.933  4414 24256 W MediaProvider: isAppCloneUserPair for user 0: false
09-13 18:21:58.955  4414 24256 I MediaProvider: Open with lower FS for /storage/emulated/0/Download/apatch_patched_10763_0.10.7_hlub.img. Uid: 10287
09-13 18:21:58.968 19025 19025 D PatchViewModel:  Write patched boot.img failed
09-13 18:21:58.969 19025 19025 D PatchViewModel: ****************************

Kernelconfig:
CONFIG_KALLSYMS=y
CONFIG_KALLSYMS_ALL=y
CONFIG_KALLSYMS_BASE_RELATIVE=y

### Reproduce method/复现方法

Try to flash APatch

### Expected behavior/预期行为

It should work

### Actual behaviour /实际行为

Can't patch the bootimage

### Device Name/设备名称

Xiaomi Mi 10T Pro 5G

### OS Version/系统版本

Voltage OS 3.2 (A14)

### APatch Version/APatch 版本

10763-166-gdcb3698 (10929)

### Kernel Version/内核版本

4.19.246

### KernelPatch Version/KernelPatch 版本

10763-166-gdcb3698 (10929)

### Additional context/其他信息

[apatch.log](https://github.com/user-attachments/files/16997400/apatch.log)
[boot.zip](https://github.com/user-attachments/files/16997429/boot.zip)

Niemandausduisburg, 2024-09-13:
I found the issue. A commit in KernelSource broke it.

Niemandausduisburg, 2024-09-13:
Now stuck on bootlogo

Hackerboy001, 2024-09-25:
I'm facing same problem

zeta96, 2024-10-14:
till now cant find what is it??

zeta96, 2024-10-18:
same here, after patching kernel, bootloop is common factor for now, kallsyms, kallsyms all, and debug kernel are enabled

bmax121, 2024-10-19:
try latest CI version

zeta96, 2024-10-19:
> try latest CI version

is that trrue apatch requrement debug kernel to enable like this:

menu "APatch Support"

config APATCH_SUPPORT

    bool "Enable Basic APatch support"

    default y

    help

            This helps you to enable requirements for APatch.

    select DEBUG_KERNEL

    select KALLSYMS
    
    select KALLSYMS_ALL
    
    select SYSFS_SYSCALL
    
    select SIGNALFD
    
    select TIMERFD
    
    select EVENTFD
    
    select BUILD_ARM64_APPENDED_DTB_IMAGE
    
    select OVERLAY_FS
    
config APATCH_FIX_MODULES

    tristate "Fix kernel modules in APatch"

    default y if APATCH_SUPPORT

    depends on APATCH_SUPPORT

    help
           Enable load modules.

    select MODULES
    
    select MODULE_FORCE_LOAD
    
    select MODULE_UNLOAD
    
    select MODULE_FORCE_UNLOAD
    
    select MODVERSIONS

config APATCH_CUSTOMS

    depends on APATCH_SUPPORT
    
    bool " APatch LocalVersion"
    
    default y if APATCH_SUPPORT
    
    help
            Add APatch to your kernel.
        
endmenu

zeta96, 2024-10-20:
> try latest CI version

i got force close with latest ci

Rev4N1, 2024-10-26:
> i got force close with latest ci

Same problem

EchoMike1776, 2025-01-26:
I am having the same issue.

closed 2025-03-09 by Admirepowered