bmax121/APatch #668: Apatch在修补的时候告诉我“patch kernel error：139“

iesaka22333, 2024-09-14:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

我知道我该去KernelPatch提问这个问题（我也确实在那里问了）但是我不确定这个错误代码一样的东西来自于谁，于是我在这也问一下，如果它与这个条目是不相干的，很抱歉，请关闭它

### Reproduce method/复现方法

将Pixel 7 Pro的Android15 Beta AP41.240823.009（或许从七月份某个版本？）的boot.img添加到apatch中修补

### Expected behavior/预期行为

修补成功

### Actual behaviour /实际行为

告诉我patch kernel error：139并输出一个无法启动的boot镜像

### Screenshots/截图

[image]

### Logs/日志

如截图

### Device Name/设备名称

Pixel 7 Pro

### OS Version/系统版本

cheetah_beta-ota-ap41.240823.009-971ac562

### APatch Version/APatch 版本

10928（更新的也试过）

### Kernel Version/内核版本

5.10.214-android13-4-00002-g4c868a6e89f5-ab12 097184

### KernelPatch Version/KernelPatch 版本

0.11.0-pre

### Additional context/其他信息

[boot.zip](https://github.com/user-attachments/files/17003223/boot.zip)
未修改的boot文件

1q23lyc45, 2024-09-15:
代码139是KernelPatch的kptool执行时崩溃了，不是APatch的问题。
可以试试用win10系统电脑手动修补(不要用WSL容器 不然大概率会报错11 Segmentation Fault)，参考这个文档 https://github.com/AndroidPatch/APatchDocs/blob/main/docs/zh_CN/install.md
```
kptools-x86_64-win.exe -p --image kernel-b --skey "YourKey" --kpimg kpimg-android --out kernel
```
不保证修补完能开机，可以试试。

iesaka22333, 2024-09-15:
> 代码139是KernelPatch的kptool执行时崩溃了，不是APatch的问题。
> 可以试试用win10系统电脑手动修补(不要用WSL容器 不然大概率会报错11 Segmentation Fault)，参考这个文档 https://github.com/AndroidPatch/APatchDocs/blob/main/docs/zh_CN/install.md 另外请使用最新的[0.11.1-dev](https://github.com/bmax121/KernelPatch/releases/tag/0.11.1-dev)
> ```
> kptools-x86_64-win.exe -p --image kernel-b --skey "你的超级密钥" --kpimg kpimg-android --out kernel
> ```
> 不保证修补完能开机，可以试试。

谢谢你的建议，我此前使用相同的KernelPatch0.1.1.0-pre版本在Linux环境中执行过修补，看上去没有报错，但刷入后它并不能正常工作。
既然不是apatch的报错那我把这个议题关一下。

closed 2024-09-15 by iesaka22333

1q23lyc45, 2024-09-15:
对了你那个pre版本从哪里下载的，能提供我一份吗？
我的手机Vivo Y52s 的 Funtouch OS 10 PD2057_A_1.16.14 版本和你这个报错一样，只有FuntouchOS_10_PD2057_A_1.7x的可以修补成功不报错，但是1.7.x底包刷gsi有问题卡第二屏，还是想用1.16.14的底包

iesaka22333, 2024-09-15:
> 对了你那个pre版本从哪里下载的，能提供我一份吗？
> 我的手机Vivo Y52s 的 Funtouch OS 10 PD2057_A_1.16.14 版本和你这个报错一样，只有FuntouchOS_10_PD2057_A_1.7x的可以修补成功不报错，但是1.7.x底包刷gsi有问题卡第二屏，还是想用1.16.14的底包

去KernelPatch 的Release 里有啊

1q23lyc45, 2024-09-15:
[image] ??????......???

iesaka22333, 2024-09-15:
> 对了你那个pre版本从哪里下载的，能提供我一份吗？
> 我的手机Vivo Y52s 的 Funtouch OS 10 PD2057_A_1.16.14 版本和你这个报错一样，只有FuntouchOS_10_PD2057_A_1.7x的可以修补成功不报错，但是1.7.x底包刷gsi有问题卡第二屏，还是想用1.16.14的底包

我脑子不是很好用，打错字了💦💦💦