bmax121/APatch #677: OnePlus 11 Using System-level EasyShare Causes System Crash and Restart

Mmz333414, 2024-09-21:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

一加11在coloros c13.1.196版本，使用系统级别的互传功能导致系统卡住崩溃重启
In ColorOS c13.1.196 version, OnePlus 11 uses the system level EasyShare function, causing the system to get stuck, crash and restart

### Reproduce method/复现方法

一加11机型刷入apatch之后，使用系统级互传功能导致重启，magisk模块为zygisknext和lspmod1.9.3以及trickysyore
The system level EasyShare function is used to restart the 1 + 11 model after it is brushed into the apatch，The Magisk modules are zygisknext and lspmod 1.9.3 and trickystore

### Expected behavior/预期行为

系统崩溃，重启
System crash, restart

### Actual behaviour /实际行为

手机系统崩溃，自动重启
使用系统自带的互传会出现
Phone system crashes, auto restart
Use the built-in EasyShare of the system will appear

### Screenshots/截图

无

### Logs/日志

在最低栏已经上传从apatch里面提取的log
The log extracted from apatch has been uploaded in the lowest column

### Device Name/设备名称

一加11 Oplus11

### OS Version/系统版本

安卓13 Android 13

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

5.15.74

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

[APatch_bugreport_2024-09-21_11_25.tar.gz](https://github.com/user-attachments/files/17082670/APatch_bugreport_2024-09-21_11_25.tar.gz)

OpM3TA, 2024-11-28:
I don't feel like making a separate issue for this but I recently switched from magisk to a patch, and I'm assuming that's the issue my phone now crashes pretty frequently I've had some system tombstone crash logs for zygote processes, usap64, and system init. I don't have 30 modules like some people although I did want to write a few of my own Colonel modules I'm a little hesitant now to even waste my time setting up the environment if my phone's going to keep crashing it's at a point where I kind of wish I had just stuck with a stock operating system on the phone because these random phone crashes and restarts are kind of making me uneasy I was honestly under the impression that this was a kind of bare metal root option and that perhaps there would be less opportunities for crashes but honestly it just seems like that may not be the case

Mmz333414, 2024-12-05:
Thank you for your reply. At present, my mobile phone is in a stable state, and there is no system crash problem. However, because apatch is not compatible with some of the software I use, I switched back to magisk!
Now, there have been several secondary restart problems by chance
Best, wish apatch better and better!

获取Outlook for Android<https://aka.ms/AAb9ysg>

closed 2025-03-09 by Admirepowered