bmax121/APatch #688: 一加12 最新版系统使用apatch root失败，但能使用KernelSU

xmhwws, 2024-09-27:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

一加12 最新版系统使用apatch失败，但能使用KernelSU
apatch 修补boot.img，刷入系统，重启，没有root权限

### Reproduce method/复现方法

正常刷入流程

### Expected behavior/预期行为

有root权限

### Actual behaviour /实际行为

apatch root失败

### Screenshots/截图

[image]

### Device Name/设备名称

一加

### OS Version/系统版本

14

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

6.1.57-android14-11-o-ga23ceff5d55d

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

系统包全名：一加OnePlus 12 ColorOS PJD110_14.0.0.850(CN01) A.78 全量包
系统包下载链接：https://gauss-otacostmanual-cn.allawnfs.com/remove-ac7389cb42fd8ee613c1f6c866e674c1/component-ota/24/09/12/f1309635171444b79af633777ac91481.zip

[APatch_bugreport_2024-09-27_18_06.tar.gz](https://github.com/user-attachments/files/17162588/APatch_bugreport_2024-09-27_18_06.tar.gz)

a1489632, 2024-09-27:
> ### APatch Version/APatch 版本
> 
> 10763
> ### Kernel Version/内核版本
> 
> 6.1.57-android14-11-o-ga23ceff5d55d
> ### KernelPatch Version/KernelPatch 版本
> 
> 0.10.7

请先使用最新ci版本重新修补测试，你用的是release版本，大概率不兼容
https://github.com/bmax121/APatch/actions/runs/11013265114

或者刷个gki

xmhwws, 2024-09-28:
> 请先使用最新ci版本重新修补测试，你用的是release版本，大概率不兼容 https://github.com/bmax121/APatch/actions/runs/11013265114
> 

使用最新ci版本patch boot.img，刷入，没有用，还是没有root权限

[image]

[image]

日志：
[APatch_bugreport_2024-09-28_16_46.tar.gz](https://github.com/user-attachments/files/17173803/APatch_bugreport_2024-09-28_16_46.tar.gz)

bmax121, 2024-10-19:
现在最新版呢

xmhwws, 2024-10-19:
> 现在最新版呢

@bmax121 大佬，最新版还是不行，使用的是：https://github.com/bmax121/APatch/actions/runs/11415983275

[image]

我的环境是kernelsu root后的（安装了zygisk插件+几个magisk模块），然后我先本地安装全量系统包，也就是上面的链接：
系统包全名：一加OnePlus 12 ColorOS PJD110_14.0.0.850(CN01) A.78 全量包
系统包下载链接：https://gauss-otacostmanual-cn.allawnfs.com/remove-ac7389cb42fd8ee613c1f6c866e674c1/component-ota/24/09/12/f1309635171444b79af633777ac91481.zip

安装系统后，重启手机，此时，kernelsu 没有root权限了。
安装最新版apatch app，用它patch boot.img，修补过程日志：
```
****************************
****************************
APatch Boot Image Patcher
****************************
- Patching kernel
+ ./kptools -p -i kernel.ori -S 超级密码人工打码 -k kpimg -o kernel
[+] kernel image_size: 0x02157a00
[+] kernel uefi header: true
[+] kernel load_offset: 0x00000000
[+] kernel kernel_size: 0x02300000
[+] kernel page_shift: 12
[+] new kernel image ...
[+] linux_banner 1: Linux version 6.1.57-android14-11-o-ga23ceff5d55d (build-user@build-host) (Android (10087095, +pgo, +bolt, +lto, -mlgo, based on r487747c) clang version 17.0.2 (https://android.googlesource.com/toolchain/llvm-project d9f89f4d16663d5012e5c09495f3b30ece3d2362), LLD 17.0.2) #1 SMP PREEMPT Mon Aug 19 11:22:00 UTC 2024
[+] linux_banner offset: 0xff0108
[+] linux_banner 2: Linux version 6.1.57-android14-11-o-ga23ceff5d55d (build-user@build-host) (Android (10087095, +pgo, +bolt, +lto, -mlgo, based on r487747c) clang version 17.0.2 (https://android.googlesource.com/toolchain/llvm-project d9f89f4d16663d5012e5c09495f3b30ece3d2362), LLD 17.0.2) #1 SMP PREEMPT Mon Aug 19 11:22:00 UTC 2024
[+] linux_banner offset: 0x1257768
[+] kernel version major: 6, minor: 1, patch: 57
[+] kallsyms_token_table offset: 0x01465720
[+] endian: little
[+] kallsyms_token_index offset: 0x01465aa8
[+] arm64 relocation kernel_va: 0xffffffffffffffff
[?] can't find arm64 relocation table
[-] /home/runner/work/KernelPatch/KernelPatch/tools/kallsym.c:490/find_markers_1(); find kallsyms_markers error
[+] kallsyms_markers range: [0x014040b0, 0x014046c4), count: 0x00000184
[+] approximate kallsyms_offsets range: [0x012578a4, 0x012b8904) count: 0x00018418
[+] kallsyms_names offset: 0x012b8910
[?] can't find kallsyms_num_syms, try: 0x0001840e
[+] names table linux_banner index: 0x0000f96e
[+] linux_banner index: 1
[+] kallsyms_offsets offset: 0x012578a8
[+] pid_vnr: type: T, offset: 0x000e3ce8
[+] pid_vnr verfied sp_el0, insn: 0xd5384108
[+] layout kimg: 0x0,0x2157a00, kpimg: 0x2158000,0x2c110, extra: 0x2184110,0x80, end: 0x2184190, start: 0x2300000
[+] kpimg version: b01
[+] kpimg compile time: 01:18:26 Oct  7 2024
[+] kpimg config: android, release
[+] tcp_init_sock: type: T, offset: 0x00393470
[+] map_start: 0x393470, max_size: 0x800
[+] kallsyms_lookup_name: type: T, offset: 0x00532880
[?] no symbol: printk
[+] _printk: type: T, offset: 0x00722850
[+] memblock_reserve: type: T, offset: 0x00538610
[+] memblock_free: type: T, offset: 0x005384a0
[+] memblock_mark_nomap: type: T, offset: 0x0053890c
[+] memblock_phys_alloc_try_nid: type: T, offset: 0x01b61540
[?] no symbol: memblock_virt_alloc_try_nid
[+] memblock_alloc_try_nid: type: T, offset: 0x01b617e0
[+] panic: type: T, offset: 0x0076b588
[+] rest_init: type: T, offset: 0x00fb5968
[+] kernel_init: type: t, offset: 0x00fb5a4c
[+] report_cfi_failure: type: T, offset: 0x00bbf928
[?] no symbol: __cfi_slowpath_diag
[?] no symbol: __cfi_slowpath
[+] copy_process: type: t, offset: 0x0030b1a4
[+] avc_denied: type: t, offset: 0x0003f334
[+] slow_avc_audit: type: T, offset: 0x00195d18
[+] input_handle_event: type: T, offset: 0x00564cb0
[+] root superkey hash: df1468c1d493537b9a7c0dcd4b12b0aa5b5d968cc3ce7f3d20a550f817a6a243
[+] paging_init: type: T, offset: 0x01b602e4
[+] patch done: kernel
+ patch_rc=0
+ set +x
- Repacking boot image
- Successfully Patched!
Output file is written to 
/storage/emulated/0/Download/apatch_patched_10971_0.11.1-dev_czen.img
****************************

```

然后命令行patch：
[image]

手机自动重启，发现apatch还是没有root权限，也就是失败了

apatch导出日志：
[APatch_bugreport_2024-10-20_00_17.tar.gz](https://github.com/user-attachments/files/17446360/APatch_bugreport_2024-10-20_00_17.tar.gz)

bmax121, 2024-10-20:
别用中文秘药

xmhwws, 2024-10-20:
> 别用中文密钥

@bmax121 超级密钥是数字+英文的，日志里面我进行人工打码了：
`+ ./kptools -p -i kernel.ori -S 超级密码人工打码 -k kpimg -o kernel`

issue重新reopen吧

bmax121, 2024-10-21:
0.10.7 可以吗

xmhwws, 2024-10-21:
> 0.10.7 可以吗

@bmax121 大佬，截至目前为止，最新版release和ci版本，都不行
ci版本链接：https://github.com/bmax121/APatch/actions/runs/11415983275

bmax121, 2024-10-21:
上传这三个东西
1. 原始 boot，
2. ksu 修补的 boot，
3. ksu 开机后
echo 1 > /proc/sys/kernel/kptr_restrict
cat /proc/kallsyms

xmhwws, 2024-10-21:
@bmax121 
[image]

注意：现在，ksu 修补的是init_boot.img，不是boot.img!!! 只需要刷入init_boot.img

相关文件已经上传链接：
https://drive.google.com/file/d/1rJr1JhRQEa1CcAxOpzi-SLKTBWWVzBPM/view?usp=sharing

bmax121, 2024-10-23:
看不到日志，试试 patch KSU 的 boot.img, 然后启动后用 root 去查看下 dmesg

xmhwws, 2024-10-23:
> 看不到日志，试试 patch KSU 的 boot.img, 然后启动后用 root 去查看下 dmesg

@bmax121 你说的日志是指kallsyms吗？我上传到压缩包里面了
[image]

bmax121, 2024-10-23:
不是，内核日志，dmesg，开机后快点拿，不然会被刷掉

suprohub, 2024-11-23:
да госпоби общайтесь уже на нормальном языке

xmhwws, 2024-11-24:
> 不是，内核日志，dmesg，开机后快点拿，不然会被刷掉

@bmax121 
环境是上面提到的，软件版本也没变。在ksu有效的情况下，对系统包里面的boot.img用apatch来patch，刷入。
开机后，没解锁前，立即执行下面命令：
`adb shell su -c "dmesg" > test.log`
得到日志：
[test.log](https://github.com/user-attachments/files/17893577/test.log)

发现apatch还是没有root权限

FirstMelody, 2024-12-16:
> > 不是，内核日志，dmesg，开机后快点拿，不然会被刷掉
> 
> @bmax121 环境是上面提到的，软件版本也没变。在ksu有效的情况下，对系统包里面的boot.img用apatch来patch，刷入。 开机后，没解锁前，立即执行下面命令： `adb shell su -c "dmesg" > test.log` 得到日志： [test.log](https://github.com/user-attachments/files/17893577/test.log)
> 
> 发现apatch还是没有root权限

apatch是不会自动获取root权限的 你确定输入了超级密钥吗？如果提示鉴权失败，并且密码正确才确定是刷入失败
我的一加12使用apatch没有问题 可以正常使用

xmhwws, 2024-12-17:
> apatch是不会自动获取root权限的你确定输入了超级钥匙吗？如果提示鉴权失败，并且密码正确才确定是刷入失败 我的一加12使用apatch没有问题正常可以使用

我刷入后类似下面图片：
[image]

刷入应该是没有问题的，我也没搞懂为什么我的设备会失败，目前只能用ksu

FirstMelody, 2024-12-18:
> 我刷入后类似下面图片： [...]

你要点击超级密钥按钮 然后输入你patch时候使用的超级密钥..... 他不会自动就跳出来的

closed 2024-12-24 by xmhwws