bmax121/APatch #695: su隐藏问题

Ethanol-Z, 2024-10-03:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

应用在未授予超级用户的情况下检测到了root

### Reproduce method/复现方法

1. 打开Apatch管理器
2. 确认某应用未授予超级用户（以及开启“排除模块”功能）
3. 打开该应用

### Expected behavior/预期行为

正常运行且不认为在root环境中

### Actual behaviour /实际行为

应用声称设备已root

### Screenshots/截图

[image]
[image]

### Logs/日志

[APatch_bugreport_2024-10-04_00_17.tar.gz](https://github.com/user-attachments/files/17247153/APatch_bugreport_2024-10-04_00_17.tar.gz)

### Device Name/设备名称

OnePlus PJD110 （一加12）

### OS Version/系统版本

14

### APatch Version/APatch 版本

10941

### Kernel Version/内核版本

6.1.57

### KernelPatch Version/KernelPatch 版本

0.11.1

### Additional context/其他信息

此外，还情况是进行了一次OTA之后发生的，同时在安装OTA之前更新了Apatch管理器，升级了系统补丁。
目前我有以下几种推测：
- ColorOS的新机制导致su对任何应用开放（可被检测到）
- 管理器的bug
- 重新修补镜像时的bug

目前只能通过修改su位置来绕过检测

Ethanol-Z, 2024-10-03:
跟进一下，重新修补一下仍然有这个问题，排除对修补映像时产生问题这种可能

whao123yahoo, 2024-10-12:
操作有问题吧

closed 2024-10-19 by bmax121