bmax121/APatch #706: root detected by Chinatax 电子税务局

MaverickYing, 2024-10-08:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

root detected
app: 电子税务局 cn.gov.chinatax.gt4.app
version: 1.1.1(330)
[image]

### Reproduce method/复现方法

1 install 电子税务局
2 click app
3 ctd

### Expected behavior/预期行为

used normally

### Actual behaviour /实际行为

crash with notification

### Screenshots/截图

[image]

### Logs/日志

[APatch_bugreport_2024-10-08_14_56.tar.gz](https://github.com/user-attachments/files/17289262/APatch_bugreport_2024-10-08_14_56.tar.gz)

### Device Name/设备名称

Oneplus 12

### OS Version/系统版本

Coloros 12 PJD110_14.0.0.850(CN01U130P00)

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

6.1.57

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

[image]

Ai686Leo, 2024-10-10:
加排除列表并在模块zygisk设置遵守排除列表，然后把应用数据清空一下试试？

MaverickYing, 2024-10-11:
找到问题了 是lsposed的问题

closed 2024-10-11 by MaverickYing

FrankieZJWZ, 2025-02-11:
是不是用原版的LSP 就可以？