bmax121/APatch #711: APM lost

3306780700, 2024-10-11:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

重启后，先前安装的模块全部丢失
原来安装的模块全部不启动，但数据还在

### Reproduce method/复现方法

- 1. 重启
- 2. 模块没了

### Expected behavior/预期行为

重启后应当加载模块，而不是彻底移除模块

### Actual behaviour /实际行为

模块被删除了，但数据还在

### Logs/日志

日志未正确保存

### Device Name/设备名称

Xiaomi Mi 11 Lite 5G

### OS Version/系统版本

LineageOS 21

### APatch Version/APatch 版本

10946

### Kernel Version/内核版本

5.4

### KernelPatch Version/KernelPatch 版本

0.11.1

### Additional context/其他信息

导出的日志大小为0B

flfwpl, 2024-10-11:
me too

poco f3
android 13 (33)
apatch apatch_10763-186-g29a691b_10949
kernel 4.19.157-perf-g92c089fc2d37
kernel patch 0.11.1

log reboot after install module
[APatch_bugreport_2024-10-12_00_06.tar.gz](https://github.com/user-attachments/files/17345688/APatch_bugreport_2024-10-12_00_06.tar.gz)

log the next reboot (module gone)
[APatch_bugreport_2024-10-12_00_10.tar.gz](https://github.com/user-attachments/files/17345697/APatch_bugreport_2024-10-12_00_10.tar.gz)

google-mirror, 2024-10-12:
不知道作者为啥要挂载这个, 多此一举还增加检测点
```
/dev/block/loop41 on /data/adb/modules type ext4 (rw,seclabel,relatime)
```

Ai686Leo, 2024-10-13:
请问最新的测试版是在哪里下的🙏

bocchi810, 2024-10-15:
CI 版受害者（

joeyoropesa-dev, 2024-10-21:
+1 and also it fails to install new zip modules. Instead they are installed in modules_update directory and never moved to right directory /data/adb/modules (only unpacked zip is in /data/adb/modules but not touched at all so customize.sh and other files are not properly executed and bins, libs installed)

@bmax121 (In version 10982 too)

Admirepowered, 2024-12-04:
https://github.com/bmax121/APatch/commit/b843480c4f56b6190add41366e3eb7148ebc9b87  check commit message

closed 2024-12-04 by Admirepowered