bmax121/APatch #725: Error al actualizar Apatch última versión

Darlin19902015, 2024-10-18:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [ ] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

[image]

### Reproduce method/复现方法

1 actualizar Apatch 
2 backup boot.imag recovery twrp
3 abrir apatch y subir boot.imag
4 error

### Expected behavior/预期行为

[image]

### Actual behaviour /实际行为

Al momento de instalar el boot.ima aparece este error.

### Device Name/设备名称

Poco x3 pro 

### OS Version/系统版本

Android 12 Miui 13

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

4.14.180

### KernelPatch Version/KernelPatch 版本

10763

### Additional context/其他信息

[image]

Darlin19902015, 2024-10-20:
Dejo un video del error 

https://github.com/user-attachments/assets/d36b77dd-91a4-464b-85ce-49faf1cc9d15

marat2509, 2024-10-27:
1. Try latest nightly version
    > [x] I am using latest CI version of APatch/我正在使用最新 CI 版本

2. Try boot.img from your ROM zip (for Recovery or Fastboot)

closed 2025-03-09 by Admirepowered