bmax121/APatch #726: AP modules

Ussr775, 2024-10-19:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Installed modules are not saved after the second reboot of the device after installing the modules. 

### Reproduce method/复现方法

1. Section "AP Modules". 
2. The button "install AP modules". 
3. Module selection. 
4. Installation of the modules. 
5. Restart the device. 
6. Section "AP Modules" (modules are installed and working).
7. Restart the device. 
8. Section "AP Modules" (modules are missing).

### Expected behavior/预期行为

The modules will be saved after the second reboot of the device and subsequent reboots will also be saved.
[image]

### Actual behaviour /实际行为

The modules are missing after the second reboot.
[image]

### Screenshots/截图

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Device Name/设备名称

Realme 9 Pro

### OS Version/系统版本

LineageOS 21

### APatch Version/APatch 版本

APatch_10763-193-g708c703_10956-release-signed

### Kernel Version/内核版本

5.4.280-qgki-ge408f03a5c42 #1 Mon Oct 14 09:06:10 UTC 2024

### KernelPatch Version/KernelPatch 版本

0.11.1

### Additional context/其他信息

After installing the modules on the first reboot, there is no modules.img file in the /data/adb/ap folder, but there is a modules_update.img file. The modules.img and modules_update.img files are missing in the /data/adb/ap folder after the second reboot of the device and, accordingly, there are no installed modules.

wangxue415, 2024-10-19:
I encountered the same problem when I used version 10971.

bmax121, 2024-10-20:
@pomelohan 这个知道咋回事不

Ussr775, 2024-10-20:
APatch_10763-170-g535146d_10933-release-signed - this is the latest version of apatch, which this bug does not have.

QQJemRun, 2024-10-21:
all of the modules disappeared after reboot 😭

ziddey, 2024-10-22:
> APatch_10763-170-g535146d_10933-release-signed - this is the latest version of Apatch, which this bug does not have.

sadly no luck here, or there's something else that's causing all my modules to get erased. Modules did survive one initial reboot, but after configuring pixelxpert and rebooting, all modules are gone again.

google-mirror, 2024-10-22:
> all of the modules disappeared after reboot 😭

expected behavior, it is a hidden feature

QQJemRun, 2024-10-22:
> > all of the modules disappeared after reboot 😭
> 
> expected behavior, it is a hidden feature

but why do that ...

ziddey, 2024-10-22:
> sadly no luck here, or there's something else that's causing all my modules to get erased. Modules did survive one initial reboot, but after configuring pixelxpert and rebooting, all modules are [...]

ok, 10933, if I kill the com.android.systemui process, all modules will get erased on the next boot. Or if systemui "naturally" crashes, then all modules will go bye-bye...

swg0101, 2024-10-23:
Also same issue on 10976.

HpNoTiQ56, 2024-10-24:
Résolve for me on 10984

Ussr775, 2024-10-24:
> Résolve for me on 10984

Hi, everything is working now, modules are installed and saved with two or more reboots.

joaoldias, 2024-10-24:
When using module [zygist detach](https://github.com/j-hc/zygisk-detach), after applying changes in the module configurations (using the Open button), all modules will be gone in the next reboot. Can any one try and confirm the issue in the latest version?
No issues using 10933

Ussr775, 2024-10-25:
> When using module [zygist detach](https://github.com/j-hc/zygisk-detach), after applying changes in the module configurations (using the Open button), all modules will be gone in the next reboot. Can any one try and confirm the issue in the latest version? No issues using 10933

I had it happen on this version so from any module I installed. Starting with APatch_10763-214-gb843480_10977-release-signed, this problem has been fixed and many thanks to the developers for this!

antoniro, 2025-02-19:
Just updated and I am having the same issue

closed 2025-03-09 by Admirepowered