bmax121/APatch #727: Write patched boot.img failed

NotchApple1703, 2024-10-20:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [ ] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Can't patch boot.img

### Reproduce method/复现方法

- Patch boot.img

### Expected behavior/预期行为

Patch boot.img successfully

### Actual behaviour /实际行为

Write patched boot.img failed

### Screenshots/截图

[image]

### Logs/日志

[APatch_bugreport_2024-10-20_10_52.tar.gz](https://github.com/user-attachments/files/17448641/APatch_bugreport_2024-10-20_10_52.tar.gz)

### Device Name/设备名称

Xiaomi Redmi K40 (alioth)

### OS Version/系统版本

SyberiaOS 7.3

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

4.19.246

### KernelPatch Version/KernelPatch 版本

10763

### Additional context/其他信息

I also tried using Apatch nightly and CI build but it crash right away when I patch boot.img

[boot.zip](https://github.com/user-attachments/files/17448674/boot.zip)

asterlx, 2024-10-23:
same issue here

zeta96, 2024-10-23:
https://github.com/bmax121/APatch/issues/666 till now, no one fix it

marat2509, 2024-10-27:
Duplicate of https://github.com/bmax121/APatch/issues/666

Well, also, the patching issues are related to KernelPatch

closed 2024-10-27 by marat2509