bmax121/APatch #752: Unable to manage su permissions and installing modules

MarekSurek, 2024-10-28:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Sorry but unable to send log since any of the sharing methods ending with error and file has 0B.
Device realme 7 pro, EvolutionX Android 14
I'm using apatch very first time, it shows on home screen 'working' but when I go to su permissions it's infinite loading and when you try to install any module it's leaving blank screen.

### Reproduce method/复现方法

1. Open permissions tab
2. See infinite loading
3. Try to install AP module
4. See blank screen

### Expected behavior/预期行为

As a example I used tricky store and zygisk next module which should be installed
Also after going to permissions, it should display privileged apps

### Actual behaviour /实际行为

Modules installation hang on blank screen 'installing'
Permissions tab infinite loading

### Screenshots/截图

[image]
[image]
[image]
[image]

### Logs/日志

.

### Device Name/设备名称

RMX2170

### OS Version/系统版本

14

### APatch Version/APatch 版本

10994

### Kernel Version/内核版本

4.14.336

### KernelPatch Version/KernelPatch 版本

0.11.1

### Additional context/其他信息

.

MarekSurek, 2024-10-28:
I did that, even clean flashed ROM but nothing

Obnomus, 2024-10-31:
> Hmm sorry it was fixed i

> Use the latest ci which is 10995.

I don't know what does ci and Where I can find it!

Obnomus, 2024-10-31:
> Use the latest ci which is 10995.

What does ci means and where can I find it!

Obnomus, 2024-10-31:
> > > Use the latest ci which is 10995.
> > 
> > 
> > What does ci means and where can I find it!
> 
> https://github.com/bmax121/APatch/actions/runs/11566966783/artifacts/2115700040

I was having this issue, so I just switched to magisk, but I'll try this next tine when I'll clean flash.

Mr-trololol, 2024-11-23:
May i know the kernel version having this issue? I've faced this in 4.14

Obnomus, 2024-11-23:
Yep I'm on version 4.14

[image]

closed 2025-03-09 by Admirepowered