bmax121/APatch #757: APatch version: 10763-240-gef48bc6 (11003)  Brand: realme Model: RMX2072 SDK Level: 31 Time: 2024-10-31 10:45:21  Thread: Thread-8 Crash Info:  java.lang.NullPointerException: Attempt to read from field 'java.lang.Object Cz.e' on a null object reference in method 'java.lang.Object i.a()' 	at i.a(Unknown Source:447) 	at g80.run(Unknown Source:2)

1813501106, 2024-10-31:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

更新后重新鉴权，输入秘钥就闪退，现在用任何版本修补刷入后都鉴权就闪退/崩溃
[image]

### Reproduce method/复现方法

重新鉴权输入密码点击确认

### Expected behavior/预期行为

之前一切正常的

### Actual behaviour /实际行为

崩溃闪退

### Screenshots/截图

[image]

### Logs/日志

APatch version: 10763-240-gef48bc6 (11003)

Brand: realme
Model: RMX2072
SDK Level: 31
Time: 2024-10-31 10:45:21

Thread: Thread-8
Crash Info: 
java.lang.NullPointerException: Attempt to read from field 'java.lang.Object Cz.e' on a null object reference in method 'java.lang.Object i.a()'
	at i.a(Unknown Source:447)
	at g80.run(Unknown Source:2)

### Device Name/设备名称

realme x50pro 玩家版

### OS Version/系统版本

realme ui f20

### APatch Version/APatch 版本

11003

### Kernel Version/内核版本

4.19.157-perf+

### KernelPatch Version/KernelPatch 版本

未知

### Additional context/其他信息

APatch version: 10763-240-gef48bc6 (11003)

Brand: realme
Model: RMX2072
SDK Level: 31
Time: 2024-10-31 10:45:21

Thread: Thread-8
Crash Info: 
java.lang.NullPointerException: Attempt to read from field 'java.lang.Object Cz.e' on a null object reference in method 'java.lang.Object i.a()'
	at i.a(Unknown Source:447)
	at g80.run(Unknown Source:2)

HuaJiOS, 2024-10-31:
我也遇到这个bug了，错误一致，刷入旧版本boot也没办法恢复
[image]

VisionR1, 2024-10-31:
Hi, I had the same, and unfortunately not fixed by go to previous version, and can't get root.
I fixed by flash original boot.img and install Magisk and go to data/adb and delete everything and uninstall Magisk, after you can go again in APatch (older version than this with this problem).

closed 2024-11-03 by Admirepowered