bmax121/APatch #763: Issues or not with .bind_mount_enable ?

VisionR1, 2024-11-03:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

In case someone have the same.
Latest nightly build with feature .bind_mount_enable, have some issues, the modules not update, example, the PIFork when I create the script-only-mode and install it again from APatch, instead of script-only-mode, install the full version again.
And module who remove or replace system apps, when install it the integrity failed, even the basic not pass, with PIF, even with TrickyStore.
If disabled my module for remove&replace files system, the integrity work again.
In Kitsune Mask I don't have problems with this specific feature.

### Reproduce method/复现方法

1. Update a module. example from PIFork full mode to script-only-mode.

2. Install a module, who remove&replace system files.

### Expected behavior/预期行为

1. Install a module actually update.

2. Not failed the integrity, even the basic.

### Actual behaviour /实际行为

All the above.

### Screenshots/截图

[image]
cp: '/data/adb/modules/playintegrityfix/module.prop' and '/data/adb/modules/playintegrityfix/module.prop' are the same file.

This feature, is from Kitsune Mask.
[image]

### Device Name/设备名称

Xiaomi Poco X3 NFC

### OS Version/系统版本

11

### APatch Version/APatch 版本

APatch_10763-236-g9b935ac/10999

### Kernel Version/内核版本

4.14.180-perf-gae096ac9215c

### KernelPatch Version/KernelPatch 版本

0.11.1

### Additional context/其他信息

Nothing more.

VisionR1, 2024-11-14:
I found when I replace hosts in system/etc and remove apps from system/pri-apps the integrity failed, even the basic, but if I replace the bootanimation from system/media, the integrity all good.

shedowe19, 2024-11-23:
[APatch_install_APM_log_2024-11-23-16-50-10.log](https://github.com/user-attachments/files/17880365/APatch_install_APM_log_2024-11-23-16-50-10.log)
[Uploading APatch_bugreport_2024-11-23_16_50.tar.gz…]()
I have read only Filesystem, can you check the Logs?

VisionR1, 2024-11-23:
> [APatch_install_APM_log_2024-11-23-16-50-10.log](https://github.com/user-attachments/files/17880365/APatch_install_APM_log_2024-11-23-16-50-10.log) Uploading APatch_bugreport_2024-11-23_16_50.tar.gz… I have read only Filesystem, can you check the Logs?

Me too only read only.
Oh, now I have again the overlayfs mode.

shedowe19, 2024-11-23:
> > [APatch_install_APM_log_2024-11-23-16-50-10.log](https://github.com/user-attachments/files/17880365/APatch_install_APM_log_2024-11-23-16-50-10.log) Uploading APatch_bugreport_2024-11-23_16_50.tar.gz… I have read only Filesystem, can you check the Logs?
> 
> Oh, now I have again the overlayfs mode. What logs exactly?

Now the Logs are there

shedowe19, 2024-11-23:
I have only write acess after Boot 5 minutes and then are in Read Only Mode

VisionR1, 2024-11-23:
> I have only write acess after Boot 5 minutes and then are in Read Only Mode

I'm little confused, how your issue, is relative with mine? You have this issue after bind_mount_enable ?

shedowe19, 2024-11-23:
[image]
With the Module Magisk_overlayfs its worked, but not without

VisionR1, 2024-11-23:
Oh, unfortunately I don't know, maybe your kernel somehow reject the internal overlayfs, and with this module can bypass.
In my device the internal overlayfs work, expect like issue title the bind_mount_enable.
Have you write about that, in the telegram Group?

backslashxx, 2024-12-03:
heya, sorry to hijack this thread, does APatch give unmount now on bind mount mode?

VisionR1, 2024-12-03:
> heya, sorry to hijack this thread, does APatch give unmount now on bind mount mode?

I don't know exactly.
I read the change there:
https://github.com/bmax121/APatch/commit/f30f346a109ca52982b44cdd315b4ef9882006ba

You can ask in Telegram group about this.

backslashxx, 2024-12-03:
so I tested this mode with someone
for module devs, when customize.sh is running, remember that MODDIR == MODPATH on it
it doesnt prepare a temporary folder, so if your moduel is named abcd

it will straight up write files to /data/adb/modules/abcd instead of a /data/adb/modules_update/abcd

only heuristic you have is the existence of /data/adb/.bind_mount_enable

VisionR1, 2024-12-03:
> so I tested this mode with someone for module devs, when customize.sh is running, remember that MODDIR == MODPATH on it it doesnt prepare a temporary folder, so if your module is named abcd it will straight up write files to `/data/adb/modules/abcd` instead of a `/data/adb/modules_update/abcd`
> 
> this will fuckup config migrations on some modules so handle it your way.
> 
> only heuristic you have is the existence of /data/adb/.bind_mount_enable

Exactly, me to, I have done test to see what happens, I hope the devs, fix this issue.

Admirepowered, 2024-12-04:
https://github.com/bmax121/APatch/tree/new  test this branch , check it work or not https://github.com/bmax121/APatch/actions/runs/12157290232

backslashxx, 2024-12-04:
> https://github.com/bmax121/APatch/tree/new test this branch , check it work or not https://github.com/bmax121/APatch/actions/runs/12157290232

finally, I hope ksu also does the same thing.
theres no way to mitigate that journaled-loop detection-point currently
https://github.com/tiann/KernelSU/issues/2225#issuecomment-2498032113

Admirepowered, 2024-12-04:
> > https://github.com/bmax121/APatch/tree/new test this branch , check it work or not https://github.com/bmax121/APatch/actions/runs/12157290232
> 
> I hope ksu also does the same thing. theres no way to mitigate that journaled-loop detection-point currently [tiann/KernelSU#2225 (comment)](https://github.com/tiann/KernelSU/issues/2225#issuecomment-2498032113)

sry,i'm not ksu dev

VisionR1, 2024-12-04:
Now the modules update works, only the integrity even basic failed, even with PIF with a FP.

backslashxx, 2024-12-04:
@Admirepowered 
will this be added to the environment?
say 

APATCH_BIND_MOUNT=true ?

this way customize.sh can differentiate

VisionR1, 2024-12-04:
@backslashxx 
For you the integrity work? For me always failed with my above setup.

backslashxx, 2024-12-04:
> @backslashxx For you the integrity work? For me always failed with my above setup.

I dont have an apatch device :D but maybe sometime, I was just debugging other people's issue

VisionR1, 2024-12-04:
> > @backslashxx For you the integrity work? For me always failed with my above setup.
> 
> I dont have an apatch device :D but maybe sometime, I was just debugging other people's issue

Understand, what root method you have?

tuyie78, 2024-12-04:
remove tricky store,use only pif

VisionR1, 2024-12-04:
> remove tricky store,use only pif
> […](#)

For me I have APatch, because no need to spoof a FP via PIF, only need a module to spoof sensitive props, like PIFork script-only-mode, and Tricky Store.

Admirepowered, 2024-12-04:
> @Admirepowered will this be added to the environment? say
> 
> ```
> APATCH_BIND_MOUNT=true
> ```
> 
> this way customize.sh can differentiate but yeah for now
> 
> ```
> [ -f /data/adb/.bind_mount_enable ] 
> ```
> 
> works.

Will he be more convenient for module development? I don't know, just add one line to the code

backslashxx, 2024-12-04:
> Will it be more convenient for module development? I don't know, just add one line to the code

well yup so we can differentiate? I guess better wait for other module devs too, let this issue to be open for some time.

ThePedroo, 2024-12-05:
In my opinion, the first one is a better approach, as not only can better adapt in the future, in the case where APatch devs want to remove this file and turn bind mount the default, but also it looks more clean.

ZG089, 2024-12-05:
> Will it be more convenient for module development? I don't know, just add one line to the code

Yes it will be more convenient.

VisionR1, 2024-12-05:
Anyone, for you the integrity work, with bind_mount_enable? For me always failed with my above setup.

backslashxx, 2024-12-06:
very nice to see
https://github.com/bmax121/APatch/commit/84732b5a06065239c978d33485e2fad93196ac47

VisionR1, 2024-12-08:
Now with latest nightly build and .bind_mount_enable, I get this with zygisk assistant and gms in excluded list.
[image]
[image]

VisionR1, 2025-01-28:
So, possibly is not APatch issue, because i go back to previous version, who have test the integrity and new and all pass, with the same modules as i have now (remove some system apps, replace host & bootanimation, PIFork & Tricky Store), so Google have changed something to detect APatch (like Magisk already is detected) module who delete system apps or replace, like a simple hosts modules.

So, if anyone have problems with Tricky Store, integrity failed, first try check this.