bmax121/APatch #767: Exclude can't uncheck.

VisionR1, 2024-11-05:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

When exclude a app (system or normal), and if I want removed after, I can't, if check it once, after can't uncheck.

### Reproduce method/复现方法

Try to uncheck a app if it is exclude.

### Expected behavior/预期行为

Can be uncheck.

### Actual behaviour /实际行为

The above.

### Device Name/设备名称

Xiaomi Poco X3 NFC

### OS Version/系统版本

11

### APatch Version/APatch 版本

11000/APatch_10763-237-g02508ff_11000-release-signed

### Kernel Version/内核版本

4.14.180-perf-gae096ac9215c

### KernelPatch Version/KernelPatch 版本

0.11.1

### Additional context/其他信息

Nothing.

ask9027, 2024-11-06:
actually it is remove app from excluded list, but not updated to APatch Manager

VisionR1, 2024-11-06:
> actually it is remove app from excluded list, but not updated to APatch Manager

Thanks for info.
So is only visual issue in APatch manager? And when I remove the app from excluded list, it has been removed regardless of remain and It still shows me check it ?

VisionR1, 2024-11-09:
So, now noticed, after reboot all show it normal.

closed 2025-02-18 by VisionR1