bmax121/APatch #774: Stuck on MIUI logo After flashing the patched boot

Fighteros, 2024-11-08:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

I'm on miui elite light a12 miui13 after flashing the patched boot rom get stuck on miui logo, when a reflash the rom the device boots successfully, i have uninstall the mods and anything the might be the reason

### Reproduce method/复现方法

Flash boot through fastboot 
reboot to the system 
the problem occurs 
reboot to recovery 
reflash the rom 
it boots successfully 

### Expected behavior/预期行为

The rom should boot because i have used it before 

### Actual behaviour /实际行为

the rom doesn't boot

### Device Name/设备名称

POCO x3 pro

### OS Version/系统版本

miui 13 android 12

### APatch Version/APatch 版本

APatch_10763_10763-release-signed

### Kernel Version/内核版本

the rom's version

### KernelPatch Version/KernelPatch 版本

no kernal patches 

### Additional context/其他信息

[image]

marat2509, 2024-11-09:
> * **If** patch failed, root failed, or **device unable to boot after flashing the new boot.img. Please goto KernelPatch**/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问

> [x] I am using latest CI version of APatch/我正在使用最新 CI 版本

No, you are not using the latest nightly version

closed 2024-11-09 by marat2509

Fighteros, 2024-11-09:
I'm on 10763 isn't that the latest ??

marat2509, 2024-11-09:
> I'm on 10763 isn't that the latest ??

Latest nightly: https://nightly.link/bmax121/APatch/workflows/build/main/APatch