bmax121/APatch #776: how to build it

jqyp324, 2024-11-11:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

How should I compile it

### Describe the solution you'd like/描述你想要的解决方案

How should I compile it

### Describe alternatives you've considered/描述您考虑过的备选方案

How should I compile it

### Additional context/其他信息

How should I compile it

VisionR1, 2024-11-11:
You can via Android studio, or via fork the APatch project, and enable action.

Fighteros, 2024-11-15:
> You can via Android studio, or via fork the APatch project, and enable action.

Where the build apk file is released after actions is completed successfully?

Fighteros, 2024-11-21:
> You can via Android studio, or via fork the APatch project, and enable action (workflows). And after like the example. ![IMG_20241115_151540](https://private-user-images.githubusercontent.com/25982450/386615379-a28b2385-a206-4d70-a39a-96e1e7fb912d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIxODk1NTEsIm5iZiI6MTczMjE4OTI1MSwicGF0aCI6Ii8yNTk4MjQ1MC8zODY2MTUzNzktYTI4YjIzODUtYTIwNi00ZDcwLWEzOWEtOTZlMWU3ZmI5MTJkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTIxVDExNDA1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ1YmE5YmI1MGM4ZmNjMWZiMjZhZTYzZjc2OTViODNlMWIyNDhhN2VhOGU4MmI5Njk0NTdmNzQ4NGY4MWM0M2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.PU45a6ESC25AzCYyoKXzae_ME4msisE2hvehuv7BzR0) ![IMG_20241115_151750](https://private-user-images.githubusercontent.com/25982450/386615393-c8b0f3f5-4f0b-4a51-b0d6-7b6f3f71f409.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIxODk1NTEsIm5iZiI6MTczMjE4OTI1MSwicGF0aCI6Ii8yNTk4MjQ1MC8zODY2MTUzOTMtYzhiMGYzZjUtNGYwYi00YTUxLWIwZDYtN2I2ZjNmNzFmNDA5LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTIxVDExNDA1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEzNDE4MzU4NjkwMWViOWFiYWVkNzgyNWIzMzZmNzg4ZmI0MzM5ZTI1NzY3NTE4ZjU0NWJkZmU3MWQ2ZTdkYTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.2dLqOd1Ss_P0wmZpbXW-YSn0J4BAuMJnCo6IzHWSB1o)

This is only what i get when fork and run the build action from github actions 

[image]

how to find the built apk for ?

VisionR1, 2024-11-21:
Must have a APatch file (zip) and inside is the apk

Fighteros, 2024-11-21:
> Must have a APatch file (zip) and inside is the apk

Yes, where i can find this ?

VisionR1, 2024-11-21:
> > Must have a APatch file (zip) and inside is the apk
> 
> Yes, where i can find this ?

Just, now I test, and me too I have the same problem, not create the artifact.

Fighteros, 2024-11-21:
> > > Must have a APatch file (zip) and inside is the apk
> > 
> > 
> > Yes, where i can find this ?
> 
> Just, now I test, and me too I have the same problem, not create the artifact.

any idea why ?

VisionR1, 2024-11-21:
> > > > Must have a APatch file (zip) and inside is the apk
> > > 
> > > 
> > > Yes, where i can find this ?
> > 
> > 
> > Just, now I test, and me too I have the same problem, not create the artifact.
> 
> any idea why ?

Try to see what wrong, I try in the Android Studio, to see if there work.

Fighteros, 2024-11-21:
If it worked with you please let me know

VisionR1, 2024-11-21:
> If it worked with you please let me know

Ok, now I update the Android Studio to latest.

VisionR1, 2024-11-21:
About GitHub action, not working because need a sign key, and must put it in the secret/action, inside in the project settings.
And create this:
[image]

I follow this:
https://github.com/marketplace/actions/sign-android-release

IzzySoft, 2024-11-22:
It's pretty easy to build it from the commandline, if you follow the build workflow. Here's what I did (on a blank `ubuntu:jammy` image in Podman, using NDK 26.3.11579264):

```yaml
- git clone https://github.com/rust-lang/rustup.git -b 1.27.0
- export ANDROID_NDK=/opt/sdk/ndk/${PROVISIONING_NDK}
- RUST_TOOLCHAIN=$(grep rust-version apd/Cargo.toml | awk -F'"' '{print $2}')
- ${BUILD_REPO_DIR}/rustup/rustup-init.sh -y --default-toolchain $RUST_TOOLCHAIN --target aarch64-linux-android
- source $HOME/.cargo/env
- cargo install cargo-ndk
- echo -e "org.gradle.parallel=true\norg.gradle.vfs.watch=true\norg.gradle.jvmargs=-Xmx2048m\nandroid.native.buildOutput=verbose" >> gradle.properties
- sed -i 's/org.gradle.configuration-cache=true//g' gradle.properties
- ./gradlew clean assembleRelease
```

The APK file can then be found in `app/build/outputs/apk/release/APatch_*-release.apk` (and `${BUILD_REPO_DIR}` in the container was `/build/repo` here, `${PROVISIONING_NDK}` as indicated `26.3.11579264`).

@Admirepowered the APK is unfortunately not identical to the one at the tag (I'm trying to establish [Reproducible Builds](https://reproducible-builds.org/) for your app, see [Reproducible Builds, special client support and more at IzzyOnDroid](https://android.izzysoft.de/articles/named/iod-rbs-mirrors-clients)). Are you sure you've used the Rust toolchain specified in `apd/Cargo.toml` at the tag? Difference is in the libraries, which usually points to a toolchain mismatch:

```diff
  -rw-r--r--  0.0 unx  3685872 b-  1867520 defN 1981-01-01 01:01:02 831b36b1 classes.dex
- -rw-r--r--  0.0 unx  2897464 b-  1303965 defN 1981-01-01 01:01:02 1213af57 lib/arm64-v8a/libapd.so
- -rw-r--r--  0.0 unx    14888 b-     5287 defN 1981-01-01 01:01:02 505d0d81 lib/arm64-v8a/libapjni.so
+ -rw-r--r--  0.0 unx  2897776 b-  1304618 defN 1981-01-01 01:01:02 1ae3bb09 lib/arm64-v8a/libapd.so
+ -rw-r--r--  0.0 unx    14888 b-     5287 defN 1981-01-01 01:01:02 4148064b lib/arm64-v8a/libapjni.so
  -rw-r--r--  0.0 unx   154248 b-    50669 defN 1981-01-01 01:01:02 6bbad534 lib/arm64-v8a/libbootctl.so
  -rw-r--r--  0.0 unx  2150056 b-  1118378 defN 1981-01-01 01:01:02 9f778850 lib/arm64-v8a/libbusybox.so
- -rw-r--r--  0.0 unx    36312 b-    13861 defN 1981-01-01 01:01:02 60b40944 lib/arm64-v8a/libkpatch.so
- -rw-r--r--  0.0 unx    50344 b-    23181 defN 1981-01-01 01:01:02 20974a23 lib/arm64-v8a/libkptools.so
+ -rw-r--r--  0.0 unx    36312 b-    13868 defN 1981-01-01 01:01:02 eaeb98b6 lib/arm64-v8a/libkpatch.so
+ -rw-r--r--  0.0 unx   290208 b-    94101 defN 1981-01-01 01:01:02 3cd8d901 lib/arm64-v8a/libkptools.so
  -rw-r--r--  0.0 unx  1208000 b-   607533 defN 1981-01-01 01:01:02 cad7ed98 lib/arm64-v8a/libmagiskboot.so
```

Admirepowered, 2024-11-22:
> It's pretty easy to build it from the commandline, if you follow the build workflow. Here's what I did (on a blank `ubuntu:jammy` image in Podman, using NDK 26.3.11579264):
>
> ```yaml
> - git clone [...]

all the ci version is publish by github action,sry understand what you said

IzzySoft, 2024-11-22:
Thanks for your swift reply, @Admirepowered!

> all the ci version is publish by github action,sry understand what you said

But that should take the toolchain from `Cargo.toml` as well? Or where should I rather pick it from? I can of course adjust the recipe, I just need to know what to :wink:

closed 2024-11-29 by Admirepowered

IzzySoft, 2024-11-29:
@Admirepowered you just closed the issue without answering my last question, maybe you have missed it? In your [workflow](https://github.com/bmax121/APatch/blob/main/.github/workflows/build.yml#L25) I don't see where the toolchain version is evaluated. If I'm not mistaken, unless explicitly specified, rustup simply installs the latest toolchain **at the time it is run** – which is not reproducible. I don't know if I'm wrong and rustup uses the `Cargo.toml`, which is why I asked. At the tag, `Cargo.toml` has `1.77.2`. If not taken from there, rustup would today use 1.83.0, which might lead to those differences. On may 2nd, rust 1.78 was released – but that was a few hours after your APK was built, so 1.77.2 seems to match. I've also used rustup 1.27.0, which was the latest stable at the time of your build.

So that should be right? Or where might there be a missing piece? I can of course drop it if you're not interested in having RB established here. Otherwise, any help is appreciated.

Oops, I've just tried again to build on noble (Ubuntu 24, as your workflow defines `ubuntu:latest`, which at the time of the release might already have been 24.04). And what happened?

```
error: failed to compile `cargo-ndk v3.5.4`, intermediate artifacts can be found at `/tmp/cargo-install71UsjF`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.

Caused by:
  package `cargo-platform v0.1.9` cannot be built because it requires rustc 1.78 or newer, while the currently active rustc version is 1.77.2
  Try re-running cargo install with `--locked`
```

Could there be some unpinned dependencies – or the call to `cargo install` should have specified a version (if I understood correctly, that can be done specifying `--version <version>` – so to match the time of your build, `cargo install cargo-ndk --version 3.4.0` – funnily, I cannot even see a 3.5.4 at https://github.com/bbqsrc/cargo-ndk/tags but I might look at the wrong repo)? Those might have caused the diff on the previous runs then. Note that I've built from the very same tag as last time, so this error should either not have happened at all, or "back then" as well. The only difference to the last run is that this time I tried on `ubuntu:noble` instead of `ubuntu-jammy`.

IzzySoft, 2024-12-06:
@Admirepowered did you just miss my comments – or do you rather want me to drop trying for RB? No bad feelings there, I just need to know so I can stop nagging you :wink:

Admirepowered, 2024-12-06:
> @Admirepowered did you just miss my comments – or do you rather want me to drop trying for RB? No bad feelings there, I just need to know so I can stop nagging you 😉

sry,there's no need to dwell on such things,You can go to the discussions to discuss not in issues,it deviated from the topic