bmax121/APatch #806: Manager not working on Lineage OS 21/Android 14

usr577, 2024-11-26:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [ ] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

When I install apatch and then open manager, the manager says it's installed, but I cannot install modules, or patch a new file and the superuser screen keeps loading indefinitely. But the lineage kernel should be compatible since all kernel configurations required according to the instructions are set to what they should be set to.

### Reproduce method/复现方法

1. Install apatch
2. Reboot
3. Open manager
(May be specific to android 14/Lineage os 21)

### Expected behavior/预期行为

Everything should be working fine

### Actual behaviour /实际行为

Superuser screen loading infinitely, modules not able to install, new files not able to be patched.

### Logs/日志

[image]
[image]
[image]

### Device Name/设备名称

Redmi note 10 pro

### OS Version/系统版本

Lineage os 21/Android 14

### APatch Version/APatch 版本

Tried nightly and latest release

### Kernel Version/内核版本

4.13.353

### KernelPatch Version/KernelPatch 版本

0.11.1

### Additional context/其他信息

Exporting logs returned a 0B zip file, seems to be related to #283 and #400 though all proposed fixes there don't work for me. Seems to be caused by CONFIG_KALLSYMS_ALL=y even though it should be supported according to documentation.

ProtoManGG, 2025-02-27:
Has there been any progress this ? I am facing the same

technopagan, 2025-04-01:
+1 on this on LOS 21 (A14). 

Switched from FDroid version of APatch Manager to APK because it kept giving me "Update Available" although the version numbers matched. Once I switched, APAtch said that its patch had a new version available, so I patched the LOS boot.img again. Now, trying to install a module yields "Error: no such file or directory (os error 2)" - before these updates, there was no error message at all but just a failed install on any module.

technopagan, 2025-04-01:
After digging around here some more my device seems to exhibit the same issue as shown in [#283](https://github.com/bmax121/APatch/issues/283) and [#400](https://github.com/bmax121/APatch/issues/400).