bmax121/APatch #812: After embedding the kpm module, it does not appear in the list of modules

middle1, 2024-11-28:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

For some reason after installing (embedding) the kpm module, it does not appear in the module lists, although when I check with dmesg, it initializes there, but it seems like it doesn't start

### Reproduce method/复现方法

- 1. Simple install kpm module

### Expected behavior/预期行为

The module works and is displayed in the list

### Actual behaviour /实际行为

The module is not working and is not displayed in the list of modules

### Screenshots/截图

[image]
[image]

### Device Name/设备名称

Realme 6 pro

### OS Version/系统版本

Android 11

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

4.14.180-perf+

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

Nop.

VisionR1, 2024-11-28:
Have you click in start button?

middle1, 2024-11-29:
> Have you click in start button?

Yes. And in dmesg there are some limitations of the modules that have been created.

VisionR1, 2024-11-29:
Have you try latest nightly build, to see if you have the same issues ?

middle1, 2024-11-29:
> Have you try latest nightly build, to see if you have the same issues ?

give a link to this build.

VisionR1, 2024-11-29:
> > Have you try latest nightly build, to see if you have the same issues ?
> 
> give a link to this build.

Is in the front page, nightly release.
https://nightly.link/bmax121/APatch/workflows/build/main/APatch

middle1, 2024-11-29:
> > > Have you try latest nightly build, to see if you have the same issues ?
> > 
> > 
> > give a link to this build.
> 
> Is in the front page, nightly release. https://nightly.link/bmax121/APatch/workflows/build/main/APatch

No, that didn't help either. The module still doesn't appear in the list and is not visible in the dmesg module log

VisionR1, 2024-11-29:
> No, that didn't help either. The module still doesn't appear in the list and is not visible in the dmesg module log

Strange. 🤔

middle1, 2024-11-29:
> > > > > Have you try latest nightly build, to see if you have the same issues ?
> > > > 
> > > > 
> > > > give a link to this build.
> > > 
> > > 
> > > Is in the front page, nightly release. https://nightly.link/bmax121/APatch/workflows/build/main/APatch
> > 
> > 
> > No, that didn't help either. The module still doesn't appear in the list and is not visible in the dmesg module log
> 
> Strange. 🤔 Have you ask about this, in the Telegram group?

Yes, but I seem to have been ignored. My kernel appears to have full support. I thought the problem was due to Magisk being installed on top, but even after completely removing it, the problem persists.

VisionR1, 2024-11-29:
> Yes, but I seem to have been ignored. My kernel appears to have full support. I thought the problem was due to Magisk being installed on top, but even after completely removing it, the problem [...]

Understand, unfortunately I don't know about that. 😕

middle1, 2024-11-29:
> Understand, unfortunately I don't know about that. 😕

That's bad. I switched from Magisk to APatch solely because of the kernel modules...

VisionR1, 2024-11-29:
Understand, if only loaded and not embedding, work?

middle1, 2024-11-29:
> Understand, if only loaded and not embedding, work?

I just rolled back the version of the apatch app itself and the module appeared in the list... Although I didn't downgrade in boo.img. I just can't figure out if it's working or not. There are detections in Hunter and Momo...

VisionR1, 2024-11-29:
> > Understand, if only loaded and not embedding, work?
> 
> I just rolled back the version of the apatch app itself and the module appeared in the list... Although I didn't downgrade in boo.img. I just can't figure out if it's working or not. There are detections in Hunter and Momo...

Understand, some apps need to add it and in the exclude list.
And about detection.
[image]

middle1, 2024-11-29:
> > > Understand, if only loaded and not embedding, work?
> > 
> > 
> > I just rolled back the version of the apatch app itself and the module appeared in the list... Although I didn't downgrade in boo.img. I just can't figure out if it's working or not. There are detections in Hunter and Momo...
> 
> Understand, some apps need to add it and in the exclude list. And about detection. ![IMG_20241129_234011](https://private-user-images.githubusercontent.com/25982450/391190749-b3008e7c-26d5-4d9a-872a-68bf7c1fcb46.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5MTY3NzQsIm5iZiI6MTczMjkxNjQ3NCwicGF0aCI6Ii8yNTk4MjQ1MC8zOTExOTA3NDktYjMwMDhlN2MtMjZkNS00ZDlhLTg3MmEtNjhiZjdjMWZjYjQ2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTI5VDIxNDExNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQzMzBmNWQxNzA2YjI1OWE0Nzc5MDZkYjMxZTgwNDYxNTRkNjM4YWUzOGFhZmQ2MzIxOGY3MWNkOGQ3YzExMGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.rx2vFVYwjVR2XDzIjduAyZRgZmhRv-Mhy845epJaeHg)

I am hiding the root due to the fact that some games and applications will stop restricting access to themselves due to the discovery of the root. Basic safety net type reviews can be passed through zygisk modules, but zygisk next itself will then be detected too. that's why I want to hide the root at the core level.

VisionR1, 2024-11-29:
> > > > Understand, if only loaded and not embedding, work?
> > > 
> > > 
> > > I just rolled back the version of the apatch app itself and the module appeared in the list... Although I didn't downgrade in boo.img. I just can't figure out if it's working or not. There are detections in Hunter and Momo...
> > 
> > 
> > Understand, some apps need to add it and in the exclude list. And about detection. ![IMG_20241129_234011](https://private-user-images.githubusercontent.com/25982450/391190749-b3008e7c-26d5-4d9a-872a-68bf7c1fcb46.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5MTY3NzQsIm5iZiI6MTczMjkxNjQ3NCwicGF0aCI6Ii8yNTk4MjQ1MC8zOTExOTA3NDktYjMwMDhlN2MtMjZkNS00ZDlhLTg3MmEtNjhiZjdjMWZjYjQ2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTI5VDIxNDExNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQzMzBmNWQxNzA2YjI1OWE0Nzc5MDZkYjMxZTgwNDYxNTRkNjM4YWUzOGFhZmQ2MzIxOGY3MWNkOGQ3YzExMGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.rx2vFVYwjVR2XDzIjduAyZRgZmhRv-Mhy845epJaeHg)
> 
> I am hiding root access due to the fact that some games and applications will begin to restrict access to themselves due to root access detection. Basic checks of the "safety net" type can be passed through the zygisk modules, but then zygisk itself will be detected. That's why I want to hide root access at the kernel level.

Yeah, understand.
You can try and the zygisk assistant.

middle1, 2024-11-30:
> Yeah, understand. You can try and the zygisk assistant.

Strangely, the AP modules now don't start at all, although they worked on the stable version. Apparently, the problem is with the KernelPatch patches.
[image]

VisionR1, 2024-11-30:
Strange, you reboot after that?

Last can you try, is delete all data/adb, remove root and after patch and flash again the boot.img

middle1, 2024-11-30:
> Strange, you reboot after that?
> 
> Last can you try, is delete all data/adb, remove root and after patch and flash again the boot.img

Yes, I have already rebooted several times. Even trying to install the module from a different version didn't help. It's unlikely that reinstalling will help, but I'll give it a try

VisionR1, 2024-11-30:
> Yes, I have already rebooted several times. Even trying to install the module from a different version didn't help. It's unlikely that reinstalling will help, but I'll give it a try

Understand, strange 🤔 
Yeah, give it a try, maybe work.

middle1, 2024-11-30:
> Understand, strange 🤔 Yeah, give it a try, maybe work.

I managed to stabilize the operation of APatch somehow, but now I'm back to the functionality that Magisk itself could provide. The Momo app detects zygisk -_-
[image]

VisionR1, 2024-11-30:
> > > > Strange, you reboot after that?
> > > > Last can you try, is delete all data/adb, remove root and after patch and flash again the boot.img
> > > 
> > > 
> > > Yes, I have already rebooted several times. Even trying to install the module from a different version didn't help. It's unlikely that reinstalling will help, but I'll give it a try
> > 
> > 
> > Understand, strange 🤔 Yeah, give it a try, maybe work.
> 
> I managed to stabilize the operation of APatch somehow, but now I'm back to the functionality that Magisk itself could provide. The Momo app detects zygisk -_- ![Screenshot_2024-11-30-21-17-42-85_8ecd68a317969a903887707449183b6f](https://private-user-images.githubusercontent.com/62933036/391263697-66ebdf84-afe3-4469-8ad0-ab379a73d6ad.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5OTEyOTQsIm5iZiI6MTczMjk5MDk5NCwicGF0aCI6Ii82MjkzMzAzNi8zOTEyNjM2OTctNjZlYmRmODQtYWZlMy00NDY5LThhZDAtYWIzNzlhNzNkNmFkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTMwVDE4MjMxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ4MTViMTIxMTU3Y2M0NTgzZDdkOGVhNTNjNWU3ZWUxZWJlYTE0OTc4YjhjN2JmNjRhYWFhNGIxNzkzMGVlZWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Kyb4jCOXuTYZ1KJoco1GZXTQFTRRikmLGdjg7hXPlEk)

Zygisk assistant
[image]

middle1, 2024-11-30:
> > > > > Strange, you reboot after that?
> > > > > Last can you try, is delete all data/adb, remove root and after patch and flash again the boot.img
> > > > 
> > > > 
> > > > Yes, I have already rebooted several times. Even trying to install the module from a different version didn't help. It's unlikely that reinstalling will help, but I'll give it a try
> > > 
> > > 
> > > Understand, strange 🤔 Yeah, give it a try, maybe work.
> > 
> > 
> > I managed to stabilize the operation of APatch somehow, but now I'm back to the functionality that Magisk itself could provide. The Momo app detects zygisk -_- ![Screenshot_2024-11-30-21-17-42-85_8ecd68a317969a903887707449183b6f](https://private-user-images.githubusercontent.com/62933036/391263697-66ebdf84-afe3-4469-8ad0-ab379a73d6ad.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5OTEyOTQsIm5iZiI6MTczMjk5MDk5NCwicGF0aCI6Ii82MjkzMzAzNi8zOTEyNjM2OTctNjZlYmRmODQtYWZlMy00NDY5LThhZDAtYWIzNzlhNzNkNmFkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTMwVDE4MjMxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ4MTViMTIxMTU3Y2M0NTgzZDdkOGVhNTNjNWU3ZWUxZWJlYTE0OTc4YjhjN2JmNjRhYWFhNGIxNzkzMGVlZWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Kyb4jCOXuTYZ1KJoco1GZXTQFTRRikmLGdjg7hXPlEk)
> 
> Zygisk assistant ![IMG_20241129_234011](https://private-user-images.githubusercontent.com/25982450/391264446-7b1b4dba-8757-4343-9a0c-e4ba6f7dd598.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5OTQyOTksIm5iZiI6MTczMjk5Mzk5OSwicGF0aCI6Ii8yNTk4MjQ1MC8zOTEyNjQ0NDYtN2IxYjRkYmEtODc1Ny00MzQzLTlhMGMtZTRiYTZmN2RkNTk4LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTMwVDE5MTMxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVjZWI3YzM1Nzc5ODZmNTYxNGI1ZDhjYzJlNWY3Mzg1Y2ZkODFmNTE0OGFiMTA4YTU3NWFiNmRhNDk1NjFlNzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.fhbgFxeQ3nbgPhv0nnLuaEpRweUYL7QH2wnu5P66eXc)

I also installed Zygisk assistant. as I wrote below, I need exactly full root hiding.

VisionR1, 2024-11-30:
Understand, unfortunately 100% cannot guarantee any modules who hide root.
Lot of parameters can cause not full hide root.

middle1, 2024-11-30:
> Understand, unfortunately 100% cannot guarantee any modules who hide root. Lot of parameters can cause not full hide root.

This is possible specifically through kernel modules. The whole problem is with zygisk, which is easily detected, but if the same operations are performed on behalf of the kernel, it is likely that it will not be detectable. And here it's not clear whether the kernel module functionality is working for me or not.

VisionR1, 2024-11-30:
> This is possible specifically through kernel modules. The whole problem is with zygisk, which is easily detected, but if the same operations are performed on behalf of the kernel, it is likely that [...]

Yeah, I get your point.
Simple, install the native detector (I think must be add it and in the exclude list), and if see something like that, that's mean the cherish beekabo not working.
[image]

yoshimo, 2024-11-30:
Why does this need additional modules to properly hide the modifications from apps?

middle1, 2024-11-30:
> > > Understand, unfortunately 100% cannot guarantee any modules who hide root. Lot of parameters can cause not full hide root.
> > 
> > 
> > This is possible specifically through kernel modules. The whole problem is with zygisk, which is easily detected, but if the same operations are performed on behalf of the kernel, it is likely that it will not be detectable. And here it's not clear whether the kernel module functionality is working for me or not.
> 
> Yeah, I get your point. Simple, install the native detector (I think must be add it and in the exclude list), and if see something like that, that's mean the cherish beekabo not working. https://github.com/reveny/Android-Native-Root-Detector ![IMG_20241130_220215](https://private-user-images.githubusercontent.com/25982450/391278367-8a6e5247-b351-4caf-81b7-29e57f86c6ee.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5OTk3MDQsIm5iZiI6MTczMjk5OTQwNCwicGF0aCI6Ii8yNTk4MjQ1MC8zOTEyNzgzNjctOGE2ZTUyNDctYjM1MS00Y2FmLTgxYjctMjllNTdmODZjNmVlLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTMwVDIwNDMyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMxY2U2OTc3ZDI2MGIzMDNjODMyMDhkMGQ5Nzk0MTNlODQwMmMxZTlkZjE3NjMzOTFiODkzMmE2NWJiMTZjOGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.9S1GUkvp8gFBv9lofZGzHeVvCfj0N_6sDHb77PnAjWQ)

With kerlen and zygisk modules
[image]

VisionR1, 2024-11-30:
So, what module you use,  need zygisk and this is the detect ?

middle1, 2024-11-30:
> So, what module you use, need zygisk and this is the detect ?

I have zygisk next, zygisk assistant, play integrity fix, and another empty module installed, which simply injects into the necessary application and that's it. I understand that this module is the one being detected, but I need it

VisionR1, 2024-11-30:
> > So, what module you use, need zygisk and this is the detect ?
> 
> I have zygisk next, zygisk assistant, play integrity fix, and another empty module installed, which simply injects into the necessary application and that's it. I understand that this module is the one being detected, but I need it

So, now is zygisk issues, must open a issue on zygisk assistant or zygisk next, with the Info of this empty module who injects.

middle1, 2024-12-01:
> So, now is zygisk issues, must open a issue on zygisk assistant or zygisk next, with the Info of this empty module who injects.

Understood. Then I have one last question, which is why I brought up this issue. Based on my screenshot from Native Detect, is the kpm module working or not?

VisionR1, 2024-12-01:
> Understood. Then I have one last question, which is why I brought up this issue. Based on my screenshot from Native Detect, is the kpm module working or not?

You mean the cherish beekabo?

VisionR1, 2024-12-01:
Yeah, as I see on your screenshot.
Because I have do some test and without cherish beekabo have this, but after no.
So, that means works.
[image]

middle1, 2024-12-01:
> Yeah, as I see on your screenshot. Because I have do some test and without cherish beekabo have this, but after no. So, that means works. ![IMG_20241130_220215](https://private-user-images.githubusercontent.com/25982450/391324709-04a2b464-075b-4c8a-8c63-ce4f534c619c.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMwNTAxNDksIm5iZiI6MTczMzA0OTg0OSwicGF0aCI6Ii8yNTk4MjQ1MC8zOTEzMjQ3MDktMDRhMmI0NjQtMDc1Yi00YzhhLThjNjMtY2U0ZjUzNGM2MTljLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjAxVDEwNDQwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMyNmJmMTIwYTkxM2QwMWM0YjJkYmYxY2M4YjRmNjIwMTQxYTg2YWM4MWJhZDNhZjAyYTZlZjllYjY2MGI0MmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.xVjhVmE1q_-BkFxzBE2pyD2kDY2xL-r5wExF6_sQsoI)

And also, how do I view the module log? Since there is no logcat, and nothing is visible through dmesg.

VisionR1, 2024-12-01:
> And also, how do I view the module log? Since there is no logcat, and nothing is visible through dmesg.

If you mean for cherish beekabo, I don't know, this specific question can ask it in TG.

middle1, 2024-12-01:
> If you mean for cherish beekabo, I don't know, this specific question can ask it in TG.

Okay, then I consider this issue closed. Thank you very much for your help)

closed 2024-12-01 by middle1