bmax121/APatch #816: Can't use a superuser

yunmo2007, 2024-11-29:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

无法使用超级用户。
can't use a superuser.

### Reproduce method/复现方法

当我打开超级用户页面,它崩溃了。
When I open the superuser page, it crashes.

### Expected behavior/预期行为

打开超级用户页面并且授予权限
Open the Superuser page and grant permissions

### Actual behaviour /实际行为

 它崩溃了。
It crash

### Device Name/设备名称

HUAWEI MHA-AL00

### OS Version/系统版本

9(API 28)

### APatch Version/APatch 版本

11007(10763-244-g68e6ead)

### Kernel Version/内核版本

4.9.111_Kirun960_Pangu_V5

### KernelPatch Version/KernelPatch 版本

0.11.1-dev

### Additional context/其他信息

内核来源于:https://github.com/maimaiguanfan/android_kernel_huawei_hi3660
The kernel is derived from:https://github.com/maimaiguanfan/android_kernel_huawei_hi3660

yunmo2007, 2024-11-29:
[APatch_bugreport_2024-11-29_21_08.tar.gz](https://github.com/user-attachments/files/17959674/APatch_bugreport_2024-11-29_21_08.tar.gz)
[image]
[image]
[image]

yunmo2007, 2024-11-29:
> ### Please check before submitting an issue/在提交 issue 前请检查
> * [x]  I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
> * [x]  If patch failed, root failed, or [...]

[APatch_bugreport_2024-11-29_21_08.tar.gz](https://github.com/user-attachments/files/17959681/APatch_bugreport_2024-11-29_21_08.tar.gz)
[image]
[image]
[image]

xhxjsk, 2024-11-30:
> ### Please check before submitting an issue/在提交 issue 前请检查
> * [x]  I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
> * [x]  If patch failed, root failed, or [...]

Cherish Peekaboo造成的，自行解决，不是apatch的问题

yunmo2007, 2024-11-30:
> > ### Please check before submit an issue/在提交 issue 前请检查
> > 
> > * [x] I have searches the issues and have not find any relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
> > * [x] 如果修补失败、root 失败或设备在刷写新的 boot.img 后无法启动。Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
> > * [x] 我将在 APatch Manager - 设置 - 报告日志/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
> > * [x] 我知道如何重现可能不特定于我的设备的问题/我知道如何重新复现这个问题
> > 
> > ### Version requirement/版本要求
> > 
> > * [x] 我正在使用 APatch/我正在使用最新 CI 版本
> > 
> > ### describe the bug/描述 bug
> > 无法使用超级用户。不能使用超级用户。
> > ### Reproduce method/复现方法
> > 当我打开超级用户页面,它崩溃了。当我打开超级用户页面时，它崩溃了。
> > ### Expected behavior/预期行为
> > 打开超级用户页面并授予权限打开超级用户页面并授予权限
> > ### Actual behaviour / 实际行为
> > 它崩溃了。它崩溃了
> > ### 截图/screenshots
> > _无响应_
> > ### 日志/日志
> > _无响应_
> > ### Device Name/设备名称
> > 华为 MHA-AL00
> > ### OS Version/系统版本
> > 9（API 28）
> > ### APatch 版本/APatch 版本
> > 11007（10763-244-G68E6EAD）
> > ### Kernel Version/内核版本
> > 4.9.111_Kirun960_Pangu_V5
> > ### KernelPatch Version/KernelPatch 版本
> > 0.11.1-开发
> > ### Additional context/其他信息
> > 内核来源于：https://github.com/maimaiguanfan/android_kernel_huawei_hi3660 内核来源：https://github.com/maimaiguanfan/android_kernel_huawei_hi3660
> 
> Cherish Peekaboo造成的，自行解决，不是apatch的问题

感谢你的回复，但是在我的重新修补了后，依旧出现了这个问题
Thanks for the reply, but after my repatch, the issue still occurs
[APatch_bugreport_2024-11-30_19_53.tar.gz](https://github.com/user-attachments/files/17965315/APatch_bugreport_2024-11-30_19_53.tar.gz)
[image]
[image]
[image]
[image]

Admirepowered, 2025-03-09:
华为的问题，有保护

closed 2025-03-09 by Admirepowered