bmax121/APatch #824: Allow Action Button For Modules

ThePieMonster, 2024-12-03:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

Just like the Magisk app has, allow a action button for installed modules which can trigger scripts.

### Describe the solution you'd like/描述你想要的解决方案

NA

### Describe alternatives you've considered/描述您考虑过的备选方案

NA

VisionR1, 2024-12-03:
The APatch already have this feature, like PIFork when click in action take a working FP from beta.

ThePieMonster, 2024-12-03:
Is there a button that is supposed to show up or some other element?

ThePieMonster, 2024-12-03:
For example this is what I see in APatch. There is no UI button and nothing happens when you click on the module. @VisionR1 

[image]

VisionR1, 2024-12-03:
> For example this is what I see in APatch. There is no UI button and nothing happens when you click on the module. @VisionR1 [...]

You have older version, before the action.

ThePieMonster, 2024-12-05:
@VisionR1 

Is there a branch for nightly builds? Why is this not being merged into main?

VisionR1, 2024-12-05:
> @VisionR1
> 
> Is there a branch for nightly builds? Why is this not being merged into main?

In the front page.
Because is not stable.
[image]

ThePieMonster, 2024-12-05:
@VisionR1 which Nightly should I be using then?

VisionR1, 2024-12-05:
> @VisionR1 which Nightly should I be using then?

Latest work fine, at least for me.
Click Nightly release, and after the first.
[image]

ThePieMonster, 2024-12-05:
Ya that seems to crash when I attempt to select the middle icon that shows apps that have root access. @VisionR1 
Using the Pixel 6 Pro on Android 13 (OEM ROM).

ThePieMonster, 2024-12-05:
> > Ya that seems to crash when I attempt to select the middle icon that shows apps that have root access. @VisionR1 Using the Pixel 6 Pro on Android 13 (OEM ROM).
> 
> Understand, I think, this is a know issue. So, go to telegram Group and ask about this, and what version can you choose.

Not able to post in the group. @VisionR1 
[image]

VisionR1, 2024-12-05:
https://web.telegram.org/k/#@apatch_discuss

ThePieMonster, 2024-12-07:
I used this version of APatch and Peekaboo 1.5.5. 

I also I am using the PIF Fork v12. When I click action there is text on the screen but it auto closes so I am not sure whats going on behind the scenes. Before running its basic only, after its basic and device. After reboot its just basic again. @VisionR1 

https://t.me/app_process64/21

VisionR1, 2024-12-07:
> I used this version of APatch and Peekaboo 1.5.5.
>
> I also I am using the PIF Fork v12. When I click action there is text on the screen but it auto closes so I am not sure whats going on behind the [...]

This is APatch Next.
Download and converter a working FP from beta.
Some users, report with cherish beekabo, drop the integrity to basic.

VisionR1, 2024-12-07:
> @VisionR1 Should I NOT be using APatch Next?
> 
> > Download and converter a working FP from beta.
> 
> What does this mean exactly?
> 
> Just tested without KPMs (Peekaboo) and only Zygist Next & PIF Fork which grabbed a beta fingerprint. I pass basic and device but no RCS.

This mean, when click the action button in PIFork, this download and converter a random working FP from beta FP, so always up to date.

Is up to you, the APatch Next have the same core as APatch, but is adapt to work with latest Cherish beekabo.
I read and write you and in the XDA, is strange why not working the RCS.

ThePieMonster, 2024-12-07:
Ok. I will continue the conversation on the XDA thread. It sounds like its fine to have APatch Next even though I am not using Cherish. 

Are there any good root detector apps out there I can use to validate if root is found on the device?

VisionR1, 2024-12-07:
> Ok. I will continue the conversation on the XDA thread. It sounds like its fine to have APatch Next even though I am not using Cherish.
> 
> Are there any good root detector apps out there I can use to validate if root is found on the device? @VisionR1

Yeah, for feature use, if you want again the cherish beekabo.
https://github.com/reveny/Android-Native-Root-Detector
https://github.com/reveny/detection

closed 2025-03-09 by Admirepowered