bmax121/APatch #825: 应用列表获取出现故障

DBPQIU, 2024-12-04:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

apatch获取不了应用列表

### Reproduce method/复现方法

无

### Expected behavior/预期行为

无

### Actual behaviour /实际行为

无

### Screenshots/截图

https://github.com/user-attachments/assets/f41e96ac-0582-4eab-918c-b037ddd5cb18

### Logs/日志

无

### Device Name/设备名称

LGv50sThinq

### OS Version/系统版本

LineagrOS

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

4.14.352-openela-perf+

### KernelPatch Version/KernelPatch 版本

4.14.352

### Additional context/其他信息

无

VisionR1, 2024-12-04:
Unfortunately, is a know issue, the infinity superuser loading.
If you search in issues here or Telegram group, you can see lot of users report the same.

closed 2025-03-09 by Admirepowered

midozalouk, 2025-04-16:
the solution lies here 

`https://github.com/sidex15/android_kernel_lge_sm8150/commit/eaa448ec761528f7ccdf3ae998b8d24ba95a8e35`

i think it is needed to be done in apatch manager not to just edit the kernel 

here is an suggestion i made 

`https://github.com/bmax121/APatch/issues/1019`