bmax121/APatch #829: 求更新

xhb615, 2024-12-06:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

更新

### Describe the solution you'd like/描述你想要的解决方案

更新

### Describe alternatives you've considered/描述您考虑过的备选方案

更新

### Additional context/其他信息

更新

1q23lyc45, 2024-12-06:
代码一直在更新，要最新版本可以用CI版，不过因为重构，请在升级后重装全部APM模块。
https://nightly.link/bmax121/APatch/workflows/build/main/APatch.zip

xhb615, 2024-12-06:
感谢开发者，我的神QwQ，爱了爱了

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年12月06日 22:29 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [bmax121/APatch] 求更新 (Issue #829) |

代码一直在更新，要最新版本可以用CI版，不过因为重构，请在升级后重装全部APM模块。
https://nightly.link/bmax121/APatch/workflows/build/main/APatch.zip

1q23lyc45, 2024-12-06:
> 感谢开发者，我的神QwQ，爱了爱了
> […](#)

没事，另外，我不是项目的开发者。

xhb615, 2024-12-06:
这个是CI版嘛

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年12月06日 22:33 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [bmax121/APatch] 求更新 (Issue #829) |

感谢开发者，我的神QwQ，爱了爱了
…
---- 回复的原邮件 ---- | 发件人 | @.> | | 日期 | 2024年12月06日 22:29 | | 收件人 | @.> | | 抄送至 | @.>@.> | | 主题 | Re: [bmax121/APatch] 求更新 (Issue #829) | 代码一直在更新，要最新版本可以用CI版，不过因为重构，请在升级后重装全部APM模块。 https://nightly.link/bmax121/APatch/workflows/build/main/APatch.zip

没事，另外，我不是项目的开发者。

1q23lyc45, 2024-12-06:
> 这个是CI版嘛
> […](#)

是的。解压缩后的apk就是。

xhb615, 2024-12-06:
好的好的收到了，谢谢😃😃

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年12月06日 22:40 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [bmax121/APatch] 求更新 (Issue #829) |

这个是CI版嘛
…
---- 回复的原邮件 ---- | 发件人 | @.> | | 日期 | 2024年12月06日 22:33 | | 收件人 | @.> | | 抄送至 | @.>@.> | | 主题 | Re: [bmax121/APatch] 求更新 (Issue #829) | 感谢开发者，我的神QwQ，爱了爱了 … ---- 回复的原邮件 ---- | 发件人 | @.> | | 日期 | 2024年12月06日 22:29 | | 收件人 | @.> | | 抄送至 | @.>@.> | | 主题 | Re: [bmax121/APatch] 求更新 (Issue #829) | 代码一直在更新，要最新版本可以用CI版，不过因为重构，请在升级后重装全部APM模块。 https://nightly.link/bmax121/APatch/workflows/build/main/APatch.zip

是的。解压缩后的apk就是。

Ai686Leo, 2024-12-07:
开发者没收咱费没义务更新的，咱别着急，慢慢等吧

xhb615, 2024-12-07:
CI版本是不是不支持magisk框架了，有些模块不适配

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年12月07日 17:20 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [bmax121/APatch] 求更新 (Issue #829) |

开发者没收咱费没义务更新的，咱别着急，慢慢等吧

1q23lyc45, 2024-12-08:
> CI版本是不是不支持magisk框架了，有些模块不适配
> […](#)

不适配的，卸载ZygiskNext(如果你安装了)，然后安装ReZygisk试试。https://github.com/PerformanC/ReZygisk
似乎LSPosed Fork by JingMatrix https://github.com/JingMatrix/LSPosed 是能用的，然后简单的替换文件模块也可以用。

wanli-Inaba, 2024-12-11:
> 代码一直在更新，要最新版本可以用CI版。不过因为重构，升级后请重装全部APM模块。 https://nightly.link/bmax121/APatch/workflows/build/main/APatch.zip

请问CI版是支持安卓15么？

wanli-Inaba, 2024-12-11:
> 代码一直在更新，要最新版本可以用CI版。不过因为重构，升级后请重装全部APM模块。 https://nightly.link/bmax121/APatch/workflows/build/main/APatch.zip

请问CI版支持安卓15的设备么？内核6.6.30  设备一加13

1q23lyc45, 2024-12-11:
> > 代码一直在更新，要最新版本可以用CI版。不过因为重构，升级后请重装全部APM模块。 https://nightly.link/bmax121/APatch/workflows/build/main/APatch.zip
> 
> 请问CI版支持安卓15的设备么？内核6.6.30 设备一加13

理论上支持，我用的一加Ace3V，安卓14，内核6.1.57 是支持的。你可以试试，但是请备份一下原厂boot防止变砖。

wanli-Inaba, 2024-12-11:
> > > 代码一直在更新，要最新版本可以用CI版。不过因为重构，升级后请重装全部APM模块。 https://nightly.link/bmax121/APatch/workflows/build/main/APatch.zip
> > 
> > 
> > 请问CI版支持安卓15的设备么？内核6.6.30 设备一加13
> 
> 理论上支持，我用的一加Ace3V，安卓14，内核6.1.57 是支持的。你可以试试，但是请备份一下原厂boot防止变砖。

14跟6.1的话官方包也支持，现在不清楚的是15能否支持T T

closed 2025-03-09 by Admirepowered