bmax121/APatch #839: 什么时候能支持安卓15呢

wanli-Inaba, 2024-12-11:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

目前骁龙8 Elite的设备全是安卓15，无法使用APatch  而狐狸头KSU这些又无法满足使用需求

### Describe the solution you'd like/描述你想要的解决方案

请求更新兼容安卓15的版本

### Describe alternatives you've considered/描述您考虑过的备选方案

.

### Additional context/其他信息

.

ossilator, 2024-12-13:
if the translation is any good, then this is a bug report about android 15 not being supported.

i installed (via sideload) today's nightly build on a xiaomi vili device flashed with crdroid 11 (android 15) beta, and it gets as far as the manager app recognizing it after entering the superkey. but the kpimg is reported to be still 0.10.7,  and apatch 10763? i'm confused.

the manager app keeps spinning in the "superuser" tab. no apps seem to be getting root.

i also can't seem to be able to install magisk modules. the only thing the install log says is "sh: <stdin>[121]: /data/adb/apd: inaccessible or not found".

FirstMelody, 2024-12-16:
安卓15已被支持 你的需求是出厂内核android版本为15的设备支持 即内核版本6.6 这个和apatch无关 上游kernelpatch已提交相关issue

ossilator, 2024-12-16:
@FirstMelody from what i can glean, you are referring to OP. but your response puzzles me. what _is_ the issue in question? do you have a link or at least a description? and why do you presume it's not related to apatch?

FirstMelody, 2024-12-18:
> @FirstMelody from what i can glean, you are referring to OP. but your response puzzles me. what _is_ the issue in question? do you have a link or at least a description? and why do you presume it's not related to apatch?

Support to android 15 refers to the manager. which is the apatch itself. What he is complaining about is devices that using Snapdragon 8Elite 's kernel cannot be patched due to kernelpatch currently did not support kernel version 6.6.

If you are using a supported kernel version, latest ci should be working on android 15. Any other problems is not related to this

ossilator, 2024-12-18:
got it. then i'll open a separate issue soon, unless the problem magically resolves.

wanli-Inaba, 2024-12-26:
> 安卓15已被支持 你的需求是出厂内核android版本为15的设备支持 即内核版本6.6 这个和apatch无关 上游kernelpatch已提交相关issue

好的，谢谢，但是我看到kernelpatch已经8个月没有更新代码了

closed 2025-03-09 by Admirepowered