bmax121/APatch #843: Cannot remount /system

dedabrane, 2024-12-12:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

When trying to remount the /system to RW the action is denied by message "APatch is read-only".

### Reproduce method/复现方法

1. Open terminal emulator app or a file manager with remount support
2. In terminal enter su
3. issue command: mount -o rw,remount /system

### Expected behavior/预期行为

/system should be mounted in RW mode for write access

### Actual behaviour /实际行为

command denied, message is: Apatch is read-only

### Screenshots/截图

[image]

### Device Name/设备名称

motorola razr 40 ultra

### OS Version/系统版本

Android 14

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

5.10.218-android12-9-00025-g7e8d2909ab18-ab12292143

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

[APatch_bugreport_2024-12-12_10_38.tar.gz](https://github.com/user-attachments/files/18109058/APatch_bugreport_2024-12-12_10_38.tar.gz)

1q23lyc45, 2024-12-12:
This is normal, because devices with a production date greater than 2022 generally leave the factory using Android 10 or above. Android 10 and above devices use the System As Root (SAR) partition architecture, and the original /system is mounted to /, but /system is not actually mounted.

Just use the new code to solve it:
```
mount -o remount,rw /
```

dedabrane, 2024-12-12:
Ok, when I issue the new command there is no error, but...
I still cannot write to /system location or any under it...
I want to have write access to /system/etc.

In addition, a question... besides this Android 14 phone I have another one (Galaxy S10+) with Android 12 and Magisk... The first mentioned mount procedure works like a charm and I have write access to everything.

Thanks!

dedabrane, 2024-12-12:
> Ok, when I issue the new command there is no error, but... I still cannot write to /system location or any under it... I want to have write access to /system/etc.
>
> In addition, a question... besides [...]

> This is normal, because devices with a production date greater than 2022 generally leave the factory using Android 10 or above. Android 10 and above devices use the System As Root (SAR) partition [...]

Ok, when I issue the new command there is no error, but...
I still cannot write to /system location or any under it...
I want to have write access to /system/etc.

In addition, a question... besides this Android 14 phone I have another one (Galaxy S10+) with Android 12 and Magisk... The first mentioned mount procedure works like a charm and I have write access to everything.

Thanks!

1q23lyc45, 2024-12-12:
Install this module.
https://github.com/HuskyDG/magic_overlayfs/releases/download/v3.2.2/magisk-overlayfs-release.zip

dedabrane, 2024-12-13:
> Install this module. https://github.com/HuskyDG/magic_overlayfs/releases/download/v3.2.2/magisk-overlayfs-release.zip

Hi @1q23lyc45 

I installed the OverlayFS module, tried to follow instructions although there isn't much to follow, but it doesn't work as it should.
Even when I set the OVERLAY_MODE=1 nothing is RW. I also tried to manually run "su -mm -c magic_remount_rw", the script seamingly runs successfully but when I list the partitions everything is again RO.
I'm guessing that maybe this is no longer your domain but rather for OverlayFS devs...

1q23lyc45, 2024-12-13:
I'm confused, this module can be used on my phone (OnePlus Ace 3V Android 14)
Upload your `/vendor/etc/fstab.qcom` and try disable AVB.

In addition, you can also try to modify the system files of the self-made module, refer to this document. https://apatch.dev/apm-guide.html, the module format is zip.

dedabrane, 2024-12-16:
> I'm confused, this module can be used on my phone (OnePlus Ace 3V Android 14) Upload your `/vendor/etc/fstab.qcom` and try disable AVB.
> 
> In addition, you can also try to modify the system files by the self-made module, refer to this document. https://apatch.dev/apm-guide.html, the module format is zip
> 
> Besides, I found that you are not using the CI version, please use the CI version. https://nightly.link/bmax121/APatch/workflows/build/main/APatch

Ok, I updated to 11015, checked AVB and verity and verify is disabled.

Can you please guide me how to test OverlayFS and to get RW?
[fstab.zip](https://github.com/user-attachments/files/18147661/fstab.zip)

I am attaching fstab.qcom...

Thanks!

dedabrane, 2024-12-19:
This seemingly works, I can edit files and create new ones...
I've noticed that phone boots a lot slower with OFS enabled, but I'm guessing this is maybe normal since it is loading a separate set of files...
Once OFS is disabled the phone boots normal and file modification is reverted, that's normal...

closed 2024-12-19 by dedabrane