bmax121/APatch #847: Apatch showing not installed even after boot.img flashing

iamrohanbro, 2024-12-14:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

I am trying to install apatch on my LG G8X running lineageos 21. The boot image gets patched successfully but after I flash it using fastboot, apatch app shows its not installed.

### Reproduce method/复现方法

1. Patched the stock boot image using Apatch. Did not embed KPM.
2. Made sure that the boot image I am patching is not already patched by magisk.
3. Transferred the patched image to Windows PC.
4. Windows PC already has latest USB drivers & platform tools installed.
5. Flashed the boot image in both boot_a & boot_b partitions using fastboot.
6. Successfully flashed in around 2.7 seconds for each partitions.
7. Rebooted.
8. Run Apatch app.

### Expected behavior/预期行为

Apatch app shows installed status.

### Actual behaviour /实际行为

Apatch shows not installed & I am given option to install.

### Screenshots/截图

[image]
[image]
[image]
[image]

### Device Name/设备名称

LG G8X (mh2lm)

### OS Version/系统版本

14

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

4.14.352-openela-perf+

### KernelPatch Version/KernelPatch 版本

255

### Additional context/其他信息

[APatch_bugreport_2024-12-14_15_28.tar.gz](https://github.com/user-attachments/files/18135328/APatch_bugreport_2024-12-14_15_28.tar.gz)

VisionR1, 2024-12-14:
Have you enter the super key after?

marat2509, 2024-12-14:
> ### Version requirement/版本要求
> * [x]  I am using latest CI version of APatch/我正在使用最新 CI 版本

closed 2024-12-14 by marat2509

iamrohanbro, 2024-12-15:
> > ### Version requirement/版本要求
> > * [x]  I am using latest CI version of APatch/我正在使用最新 CI 版本
> 
> Try latest nightly version
> 

Kindly provide me the link, I am unable to find any nightly version.

iamrohanbro, 2024-12-15:
> > ### Version requirement/版本要求
> > * [x]  I am using latest CI version of APatch/我正在使用最新 CI 版本
> 
> Try latest nightly version
> 

Same result. With latest ci build.

VisionR1, 2024-12-15:
> > > ### Version requirement/版本要求
> > > 
> > > * [x]   I am using latest CI version of APatch/我正在使用最新 CI 版本
> > 
> > 
> > Try latest nightly version
> 
> Same result. With latest ci build.
> 
> [APatch_bugreport_2024-12-15_08_51.tar.gz](https://github.com/user-attachments/files/18138873/APatch_bugreport_2024-12-15_08_51.tar.gz)

After you flash the patch boot.img, what show you in APatch?

iamrohanbro, 2024-12-15:
> After you flash the patch boot.img, what show you in APatch?

Says installation or authentication pending. If i enter superkey it says authentication failed. But i am not putting wrong password, seriously.
Btw i was asking around in telegram & some guys said apatch wont work on openela kernels. But idk if they are right.
Thank you for paying attention to my issue.
Let me know if there's something i can do.