bmax121/APatch #852: I can't use Superuser.

DogancanYr, 2024-12-17:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [ ] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

....

### Reproduce method/复现方法

....

### Expected behavior/预期行为

....

### Actual behaviour /实际行为

I can't install modules, and the su part is always loading.

### Logs/日志

I can't even create a bug report.

### Device Name/设备名称

Xiaomi Redmi Note 9 Pro

### OS Version/系统版本

Lineage os-21.0-20241216

### APatch Version/APatch 版本

11015

### Kernel Version/内核版本

...

### KernelPatch Version/KernelPatch 版本

0.11.1

### Additional context/其他信息

APatch version: 10763-252-g9ef7a6f (11015)

Brand: Redmi
Model: Redmi Note 9 Pro
SDK Level: 34
Time: 2024-12-17 22:32:04

Thread: DefaultDispatcher-worker-2
Crash Info: 
java.io.FileNotFoundException: /data/user/0/me.bmax.apatch/cache/APatch_bugreport_2024-12-17_22_32.tar.gz: open failed: ENOENT (No such file or directory)
	at libcore.io.IoBridge.open(IoBridge.java:574)
	at java.io.FileInputStream.<init>(FileInputStream.java:179)
	at M00.j(Unknown Source:72)
	at kb.o(Unknown Source:5)
	at ao.run(Unknown Source:103)
	at j2.run(Unknown Source:35)
	at u60.run(Unknown Source:2)
	at Vk.run(Unknown Source:88)
	Suppressed: jn: [mJ@2f21df, Vc@1dd8e2c, um{Cancelling}@eaf20f5, Dispatchers.IO]
Caused by: android.system.ErrnoException: open failed: ENOENT (No such file or directory)
	at libcore.io.Linux.open(Native Method)
	at libcore.io.ForwardingOs.open(ForwardingOs.java:563)
	at libcore.io.BlockGuardOs.open(BlockGuardOs.java:274)
	at libcore.io.ForwardingOs.open(ForwardingOs.java:563)
	at android.app.ActivityThread$AndroidOs.open(ActivityThread.java:8478)
	at libcore.io.IoBridge.open(IoBridge.java:560)
	... 7 more

mian196, 2025-01-29:
did you find a solution? same thing happened with me i patch with stable and nightly same thing no Su

DogancanYr, 2025-02-13:
Also not working.
0.11.2
11039

viik420, 2025-02-14:
Same issue.

farisossandon, 2025-02-16:
Same issue, Mi 9t LineageOS 22.1

TheSingular, 2025-02-21:
> Also not working. 0.11.2 11039 Lineage os 22

If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问

excalgm, 2025-03-04:
Same for me. (Redmi Note 10 Pro)
APatch 11039, KernelPatch 0.11.2, LineageOS 22.1

DogancanYr, 2025-03-05:
> Same for me. (Redmi Note 10 Pro) APatch 11039, KernelPatch 0.11.2, LineageOS 22.1

You can use kernelsu next.

yennor, 2025-03-29:
same problem with cr droid 15 on poco x3 nfc