bmax121/APatch #873: 升级到 LineageOS 22 后刷入修补镜像无法进入系统，已排除上游问题

HelixCopex, 2025-01-04:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

系统更新后（LineageOS 21 - LineageOS 22），使用最新版本的 APatch 修补 boot.img 后刷入，依次显示 BL锁警告 、厂商 Logo 后反复重启，无法正常进入系统

### Reproduce method/复现方法

按照正常流程获取 boot.img ，修补并刷入

### Expected behavior/预期行为

安卓系统本体正常加载并进入系统

### Actual behaviour /实际行为

依次显示 BL锁警告 、厂商 Logo 后无法正常进入系统（在安卓启动动画播放环节前反复重启）

### Device Name/设备名称

SONY Xperia 1 II / pdx203 / XQ-AT72

### OS Version/系统版本

LineageOS 22.1-20250103-NIGHTLY-pdx203

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

4.19.324

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

在上一个系统版本使用 APatch 时未出现此问题。
新版本更新后内核配置要求依然检查通过（CONFIG_KALLSYMS=y and CONFIG_KALLSYMS_ALL=y）
刷入系统内提取/随本体同时发布的 boot.img 可以正常进入系统。
在当前版本系统中（LineageOS 22），我尝试通过最新的内核源码与 Kernel 0.9.5 版本合并编译后，打包为 AnyKernel3 包体通过 Lineage Recovery 刷入，可以正常进入系统，且安装模块等功能正常。
我不确定这能否说明上游代码没有问题，但是 pdx203 这部设备并未使用 GKI 内核，而 KernelSU 在 1.0.0 放弃了对非 GKI 内核设备的支持。但在上一个系统版本中（LineageOS 21）APatch 功能正常，所以来这边提 issue。

chinasoul, 2025-01-09:
小米澎湃os问题类似，miui可以用，升到澎湃刷后开机卡米。换用nightly build，不卡米能进系统，但是依旧显示未安装

xlucn, 2025-01-10:
一加8+lineageos22同症状

AE-PH, 2025-01-11:
我自己的设备在从LOS21升级到22的过程中是正常的，内核是4.4。
LOS只能在recovery下升级，我是预先处理好boot后刷入系统后重启进fastboot刷写boot的。
可能是部分机型跨版本升级的问题？
此外，是否排除了GMS包的原因？跨版本时GMS会经常忘记升级，可能会有干扰。

HelixCopex, 2025-01-13:
> 我自己的设备在从LOS21升级到22的过程中是正常的，内核是4.4。 LOS只能在recovery下升级，我是预先处理好boot后刷入系统后重启进fastboot刷写boot的。 可能是部分机型跨版本升级的问题？ 此外，是否排除了GMS包的原因？跨版本时GMS会经常忘记升级，可能会有干扰。

我升级时使用的是相同步骤。GMS按照官方的升级指南刷写过，顺序如下：

1. 进 Recovery
2. 刷写系统包本体
3. 刷写 GApps
4. 从 Rec 重启到 Fastboot
5. 刷写修补过的 boot.img

有可能是 LOS 上游源码的某些 bug，我等几个 release 后再试试，最初的这两个版本小问题不断，目前暂时用 KernelSU.Next 作为 root 方案。

strec007, 2025-01-21:
Same problem here with LineageOS for microg 22 on FP4.

cribyn, 2025-01-29:
Cannot patch LineageOS microG 22 on Oneplus 8pro, having dropped back to L OS mG 21 it patched just fine without a hitch.
Mentioning in case it helps anyone.

xiliuya, 2025-02-01:
Same problem here with LineageOS 22 on  Xiaomi alioth.

ArchGuyWu, 2025-02-06:
> 我升级时使用的是相同步骤。GMS按照官方的升级指南刷写过，顺序如下：
>
> 1. 进 Recovery
> 2. 刷写系统包本体
> 3. 刷写 GApps
> 4. 从 Rec 重启到 Fastboot
> 5. 刷写修补过的 boot.img
>
> 有可能是 LOS 上游源码的某些 bug，我等几个 release 后再试试，最初的这两个版本小问题不断，目前暂时用 KernelSU.Next 作为 root 方案。

你要用apatch ci构建版，最新正式版问题多多

xlucn, 2025-02-08:
😆 确认nightly版本APatch可使用没问题。所以是已经适配了，但是没有发布新版本？
https://nightly.link/bmax121/APatch/workflows/build/main/APatch

jiongjiongzj, 2025-02-12:
同 LOS 22 ，看到楼上突然想到的解决方法：使用在线修补，勾选 Switch Channel 以切换 Prerelease 版本的 Kernel Patch 。亲测可用。
项目文档里的在线修补网址：https://kernelpatch-on-web.pages.dev/

closed 2025-03-09 by Admirepowered