bmax121/APatch #879: Pixel Fold A16 DP2 - Kernel panic when running Flutter app or receiving file with Quick Share

Kimjio, 2025-01-09:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Kernel panic occurs when launching Flutter app or receiving file with Quick Share.

It doesn't happen with Magisk.

### Reproduce method/复现方法

1. Install [Revanced Manager](https://revanced.app/download) (or another Flutter app)
2. Open Revanced Manager
3. Kernel panic

OR

1. Receive files with Quick Share
2. Kernel panic

### Expected behavior/预期行为

App opens normally and receive the file normally

### Actual behaviour /实际行为

Kernel panic

### Device Name/设备名称

Google Pixel Fold (felix)

### OS Version/系统版本

Android 16 DP2

### APatch Version/APatch 版本

11023

### Kernel Version/内核版本

6.1.112-android14-11-g887a7e86d86c-ab12609334

### KernelPatch Version/KernelPatch 版本

0.11.1

### Additional context/其他信息

Logs when launching Flutter app
[APatch_bugreport_2025-01-09_20_25.tar.gz](https://github.com/user-attachments/files/18360781/APatch_bugreport_2025-01-09_20_25.tar.gz)

Logs when receiving file with Quick Share
[APatch_bugreport_2025-01-09_20_52.tar.gz](https://github.com/user-attachments/files/18361096/APatch_bugreport_2025-01-09_20_52.tar.gz)

victorbnl, 2025-05-20:
So I am able to reproduce the issue on a Pixel 7a, as mentioned in my issue (which I had created before I came across yours). On my side, the issue happens with Flutter apps such as ReVanced Manager but also Firefox and its forks.

By looking at the `console-ramoops-0` file you can see there is an issue with `sock_ioctl`:

```
RAMDUMP_MSG.txt:
  reset message: KP: Oops: Fatal exception: comm:1.ui PC:sock_ioctl+0x70/0x55c LR:sock_ioctl+0x430/0x55c
  UUID: d638112b-a39b-5d4e-805b-70306b7f99f8
  last kernel version: 6.1.112-android14-11-g3cc78585b764-ab12609334
  aosp kernel version: 6.1.112-android14-11-g887a7e86d86c-ab12609334
  build: google/felix_beta/felix:Baklava/BP21.241121.009/12787338:user/release-keys
  RST_STAT: 0x80 - SYSTEM_SWRESET_SYSTEM
  GSA_RESET_STATUS: 0x10 - GSA_INTERMEDIATE_RESET
  Reboot reason: 0xbaba - Kernel PANIC
  Reboot mode: 0x0 - Normal Boot
```

I haven't found any workaround nor information towards a solution yet.

KodiCraft, 2025-05-23:
This issue is happening to me as well on a Pixel 7a with stock ROM. I haven't found any solution yet

albotroz, 2025-08-04:
> So I am able to reproduce the issue on a Pixel 7a (guess this is a Pixel issue then), LineageOS 22.2 (Android 15), as mentioned in my issue (which I had created before I came across yours).
> 
> Here is a list of applications that crash, with alternatives / workarounds if you still want to use APatch.
> Application 	Alternative / workaround
> Clash Royale 	
> Finch 	
> Firefox (and its forks) 	DuckDuckGo
> Guitar Tuna 	Fender Guitar Tuner
> QuickShare 	
> ReVanced Manager 	Patch on another device
> SNCF Connect 	Use the website
> WiZ Connected 	Not tested, but try riz
> 
> It seems that all torrenting applications suffer from this bug. I have tested BiglyBT, Flud, LibreTorrent and uTorrent. On LibreTorrent, it happens only after having added a torrent. Other clients make it panic at start.
> 
> By looking at the `console-ramoops-0` file you can see there is an issue with `sock_ioctl` (addresses are always `PC:sock_ioctl+0x70/0x55c LR:sock_ioctl+0x430/0x55c` on all affected apps):
> 
> ```
> RAMDUMP_MSG.txt:
>   reset message: KP: Oops: Fatal exception: comm:1.ui PC:sock_ioctl+0x70/0x55c LR:sock_ioctl+0x430/0x55c
>   UUID: d638112b-a39b-5d4e-805b-70306b7f99f8
>   last kernel version: 6.1.112-android14-11-g3cc78585b764-ab12609334
>   aosp kernel version: 6.1.112-android14-11-g887a7e86d86c-ab12609334
>   build: google/felix_beta/felix:Baklava/BP21.241121.009/12787338:user/release-keys
>   RST_STAT: 0x80 - SYSTEM_SWRESET_SYSTEM
>   GSA_RESET_STATUS: 0x10 - GSA_INTERMEDIATE_RESET
>   Reboot reason: 0xbaba - Kernel PANIC
>   Reboot mode: 0x0 - Normal Boot
> ```
> 
> I think the issue might be that somewhere up the call stack (see backtrace in the console ramoops) a NULL or bad socket is passed to `sock_ioctl`. But I might be mistaken since kernel development is beyond my understanding.
> 
> I haven't found any workaround nor hint towards a solution yet.

I have prob. same problem with Pixel 6a
- Lineage-22.2-20250802-nightly-bluejay-signed [Android Kernel 6.6]
- APatch v11107

According to the changelog, this APatch version should officially support kernel 6.6 but apparently doesn't look so...

For me, for example, other apps (besides the ones you mentioned, like Firefox) cause my smartphone to crash.

- Octopus (learning app)
- TinyCam
- Obainium

They're not really replaceable.
However, Firefox isn't replaceable for me either.

Schweeeeeeeeeeeeeeee, 2025-08-05:
Crazy that this has been reported in January and nothing happened yet