bmax121/APatch #909: Patch kernel error: 1 (symbol.c:77/fillin_map_symbol(); no symbol memblock_alloc+ patch_rc=1)

JMVS, 2025-02-03:
### Please check before submitting an issue/在提交 issue 前请检查

- [x] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [x] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [x] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [x] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Trying to patch image causes error "[-]/home/runner/work/Kernel Patch/KernelPatch/tools/symbol.c:77/fillin_map_symbol(); no symbol memblock_alloc+ patch_rc=1" (see full log below). Device info attached.
[basic.txt](https://github.com/user-attachments/files/18644917/basic.txt)

Output of `zcat /proc/config.gz | grep -w CONFIG_KALLSYMS`
`CONFIG_KALLSYMS=y `

### Reproduce method/复现方法

1. Go to 
[image]
2. Select a boot image to Patch
3. Select boot.img extracted from original file
4. Set a SuperKey
5. Click Start

### Expected behavior/预期行为

Patched boot.img

### Actual behaviour /实际行为

No patched boot.img

Error: [-]/home/runner/work/Kernel Patch/KernelPatch/tools/symbol.c:77/fillin_map_symbol(); no symbol memblock_alloc+ patch_rc=1

### Logs/日志

[APatch Boot Image Patcher.log](https://github.com/user-attachments/files/18644903/APatch.Boot.Image.Patcher.log)

### Device Name/设备名称

POCO M3 Pro 5G

### OS Version/系统版本

MIUI Global 14.0.6 (Android 13)

### APatch Version/APatch 版本

10762 (shown on title) 10763 (shown on description and file)

### Kernel Version/内核版本

#1 SMP PREEMPT Tue Sep 19 12:46:50 UTC 2023

### KernelPatch Version/KernelPatch 版本

N/A

### Additional context/其他信息

DEVICE: camellian
SELinux: Enforcing
KernelRelease: 4.14.186-perf-gf8d872e9279c
KernelVersion: #1 SMP PREEMPT Tue Sep 19 12:46:50 UTC 2023

JMVS, 2025-02-04:
Upon reading the [README.md](https://github.com/bmax121/APatch/blob/main/README.md ) I see that I need not only CONFIG_KALLSYMS=y but also CONFIG_KALLSYMS_ALL=y. This is not what it says in the [installation guide](https://apatch.dev/install.html). Can this be the cause of the error?

[image]

[image]

JMVS, 2025-03-10:
May I ask how is completed? I don't seem to find a commit associated with it. Thanks

Admirepowered, 2025-05-13:
> Any news on this? Thanks

issue should not open on here