bmax121/APatch #923: Revanced applications are not mounted on apatch

AntonLosk, 2025-02-13:
### Please check before submitting an issue/在提交 issue 前请检查

- [x] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [x] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [x] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [x] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Revanced applications are not mounted on apatch

### Reproduce method/复现方法

Install youtube revanced

### Expected behavior/预期行为

the application must be mounted

### Actual behaviour /实际行为

not mounted

### Screenshots/截图

[image]
[image]

### Logs/日志

+ handle
+ truncate RUfOPODExuzq7sl event services before
report success
+ truncate RUfOPODExuzq7sl -Z u:r:magisk:s0 exec /data/adb/apd -s RUfOPODExuzq7sl services
/data/adb/service.d/com.google.android.youtube.sh: line 3: magisk: not found
/data/adb/service.d/com.google.android.apps.youtube.music.sh: line 3: magisk: not found
+ truncate RUfOPODExuzq7sl event services after
report success

### Device Name/设备名称

Xiaomi mi 11 lite 5g ne

### OS Version/系统版本

15

### APatch Version/APatch 版本

11039

### Kernel Version/内核版本

5.4.286

### KernelPatch Version/KernelPatch 版本

0.11.2

### Additional context/其他信息

[APatch_bugreport_2025-02-13_20_24.tar.gz](https://github.com/user-attachments/files/18786036/APatch_bugreport_2025-02-13_20_24.tar.gz)

HolyDemonz, 2025-02-15:
Same thing happens in latest stable version.

Kirka42, 2025-02-16:
Apatch/KSU rw module also stopped working. It seems to me that this problem is caused by the same error. The system partition is now read-only.

nicpadilla, 2025-04-16:
Chipping in to confirm I'm experiencing the same core issue @306bobby-android described, specifically on APatch **11039**.

My script using `mount -o bind` for ReVanced apps also fails to work effectively. This happens not just with boot scripts, but also when I run the script manually from a root terminal (which has the `u:r:magisk:s0` context).

The really confusing part is that the `mount` command itself **returns an exit code of 0**, acting like it succeeded, but checking the actual system `mount` list shows the required bind mount was **not created**.

Like others found, using the mount function directly within the ReVanced Manager app's UI *does* work correctly.

Just wanted to add for extra data: I tried forcing OverlayFS mode (via `.overlay_enable`) as mentioned in the 11039 notes, but that unfortunately didn't fix this silent mount failure from the script either.

chozian, 2025-05-29:
I am experiencing the same issue with APatch version 11075.

ziddey, 2025-05-30:
I wrote a script like:
```
#!/system/bin/sh
sh -x /data/adb/service.d/com.google.android.youtube.sh >yt.out 2>&1
```

and then checking `yt.out`, notable is:
```
+ mount -o bind /data/adb/revanced/com.google.android.youtube/base.apk
mount: can't read '/etc/fstab': No such file or directory
```

Can't speak about the issue / proper solution, but since I'm running an ssh server, and the revanced script does work to mount in an ssh session, I just made an extremely dirty service.d script to do a loopback ssh session to execute the revanced script..
```
#!/system/bin/sh
until [ "$(getprop sys.boot_completed)" = 1 ]; do sleep 3; done
ssh root@localhost /data/adb/service.d/com.google.android.youtube.sh
```

`until` borrowed from the revanced script to maybe ensure the ssh server is running

306bobby-android, 2025-05-31:
> I wrote a script like:
>
> ```
> #!/system/bin/sh
> sh -x /data/adb/service.d/com.google.android.youtube.sh >yt.out 2>&1
> ```
>
> and then checking `yt.out`, notable is:
>
> ```
> + mount -o bind [...]

Does running the script work in just a normal root terminal session, i.e termux su?

If so, the ssh loopback may not be necessary, and APatch is just running service.d too early. making sure the until is in the revanced script, and adding a longer sleep before the mount takes place

Unfortunately I cannot test, as I've moved to KernelSU Next and it works fine here 😄

fawzy46, 2025-06-05:
Same here with 11039

chozian, 2025-07-04:
Same here on 11089