bmax121/APatch #927: App crash after flash a module

emptylight370, 2025-02-14:
### Please check before submitting an issue/在提交 issue 前请检查

- [x] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [x] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [x] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [x] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

After flash PUI Themed Icon Completion Project on Oneplus ACE2V, the Apatch stop for a while and show several chown error.
Similar with chown the monochrome.png:0
Then it suddenly show the crash page.
But later I can see the module installed after restart APatch.

### Reproduce method/复现方法

1. download the latest module file from PUI group
2. flash with latest Apatch
3. wait for complete
4. see some chown error
5. the app crash

### Expected behavior/预期行为

install the module without crash

### Actual behaviour /实际行为

crash when nearly installed the module

### Logs/日志

APatch version: 11039 (11039)

Brand: OnePlus
Model: PHP110
SDK Level: 35
Time: 2025-02-14 23:04:18

Thread: main
Crash Info: 
java.lang.RuntimeException: android.os.TransactionTooLargeException: data parcel size 5066424 bytes
Bundle stats:
  androidx.lifecycle.BundlableSavedStateRegistry.key [size=5065432]
    lV:-1 [size=5064360]
      -ipdx3q [size=5058844]
      9uksvn [size=5460]
PersistableBundle stats:
  [null]
	at android.app.servertransaction.PendingTransactionActions$StopInfo.run(PendingTransactionActions.java:146)
	at android.os.Handler.handleCallback(Handler.java:959)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loopOnce(Looper.java:282)
	at android.os.Looper.loop(Looper.java:387)
	at android.app.ActivityThread.main(ActivityThread.java:9505)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:600)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1005)
Caused by: android.os.TransactionTooLargeException: data parcel size 5066424 bytes
	at android.os.BinderProxy.transactNative(Native Method)
	at android.os.BinderProxy.transact(BinderProxy.java:679)
	at android.app.IActivityClientController$Stub$Proxy.activityStopped(IActivityClientController.java:1545)
	at android.app.ActivityClient.activityStopped(ActivityClient.java:120)
	at android.app.servertransaction.PendingTransactionActions$StopInfo.run(PendingTransactionActions.java:135)
	... 8 more

### Device Name/设备名称

Oneplus ACE2V

### OS Version/系统版本

ColorOS15 Android15

### APatch Version/APatch 版本

11039

### Kernel Version/内核版本

5.10.209-android12-9-o-gdbe4d00409d9

### KernelPatch Version/KernelPatch 版本

0.11.2

### Additional context/其他信息

[APatch_bugreport_2025-02-14_23_15.tar.gz](https://github.com/user-attachments/files/18801150/APatch_bugreport_2025-02-14_23_15.tar.gz)

VisionR1, 2025-02-14:
Except crash, after if reboot the module work?

emptylight370, 2025-02-15:
It works after reboot.

VisionR1, 2025-02-15:
> It works after reboot.
> […](#)

That's is good.
Even though APatch say crash, after work.

Admirepowered, 2025-02-15:
是不是网络有问题，下载或者超时了

emptylight370, 2025-02-16:
应该不会，不记得莫奈图标模块有联网环节。可能是模块里文件数量太多了？模块里都是应用图标，大概有几千个吧，刷入时候log显示是完全卡死的，刷入到post_fs阶段之后才能正常跳转到log底部，之后就是弹出两三个chown错误。
---- Replied Message ----
***@***.***>Date2/16/2025 ***@***.***>***@***.***>,
***@***.***>SubjectRe: [bmax121/APatch] App crash after flash a module (Issue #927)
是不是网络有问题，下载或者超时了
—
是不是网络有问题，下载或者超时了

Admirepowered, 2025-02-17:
有没有模块，上传一下

emptylight370, 2025-02-17:
> 只有这个模块有问题是把，有没有模块，上传一下

https://dav.emptylight.cn/s/q2FP

commit cb684e60 (2025-03-09): Fix manager crash while installing `APModule`

Fix #927.
closed 2025-03-09 by pomelohan