bmax121/APatch #929: Abnormal Mount

Ruieaas, 2025-02-15:
### Please check before submitting an issue/在提交 issue 前请检查

- [x] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [x] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [x] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [x] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

KernelPatch 0.11.2 APatch 11039 Magic Mount
Zygisk - LSPosed 7075 npm 
MemoryDetector find 
dex2ota32 dex2ota64 
in 
/apex/com.android.art/bin/           
Problems after upgrading from APatch10933 I'm sure I didn't change the LSPosed version or change the settings   

### Reproduce method/复现方法

install LSPosed7075 npm

### Expected behavior/预期行为

MemoryDetector can't find LSPosed

### Actual behaviour /实际行为

MomeryDetector find
dex2ota32 dex2ota64 
in 
/apex/com.android.art/bin/           

### Device Name/设备名称

OnePlus11 PDB110 Salami

### OS Version/系统版本

OxygenOS14 Android14

### APatch Version/APatch 版本

APatch11039

### Kernel Version/内核版本

5.15.123 android13-8

### KernelPatch Version/KernelPatch 版本

KernelPatch0.11.2

### Additional context/其他信息

[APatch_bugreport_2025-02-15_11_03.tar.gz](https://github.com/user-attachments/files/18807615/APatch_bugreport_2025-02-15_11_03.tar.gz)

VisionR1, 2025-02-15:
I think not issue with APatch.
Can you try this? Especially action build.
https://github.com/JingMatrix/LSPosed

Ruieaas, 2025-02-15:
> 我认为 APatch 没有问题。您可以尝试这个吗？尤其是动作构建。 https://github.com/JingMatrix/LSPosed
> 
> [image]

I now think the problem stems from Magic Mount, in Magisk and KernelSU where it is used LSPosed is usually featured or detected as mounted, and with exclusion enabled MemoryDetector doesn't detect it

VisionR1, 2025-02-15:
> > 我认为 APatch 没有问题。您可以尝试这个吗？尤其是动作构建。 https://github.com/JingMatrix/LSPosed
> > [image]
> 
> I now think the problem stems from Magic Mount, in Magisk and KernelSU where it is used LSPosed is usually featured or detected as mounted, and with exclusion enabled MemoryDetector doesn't detect it

Understand.
You can try the overlays mode, to see if change anything.

Ruieaas, 2025-02-15:
> > > 我认为 APatch 没有问题。您可以尝试这个吗？尤其是动作构建。https://github.com/JingMatrix/LSPosed
> > > [image]
> > 
> > 
> > 我现在认为问题源于 Magic Mount，在使用它的 Magisk 和 KernelSU 中，LSPosed 通常被推荐或检测为已安装，并且在启用排除的情况下，MemoryDetector 不会检测到它
> 
> 理解。您可以尝试 overlays 模式，看看是否有任何更改。

No problems found in Ovrlayfs mode
Enabling APatch's exclusion module in Magic Mount mode solves the problem
I hope you can tell more people about this 😋

Ruieaas, 2025-02-15:
> > > > 我认为 APatch 没有问题。您可以尝试这个吗？尤其是动作构建。https://github.com/JingMatrix/LSPosed
> > > > [image]
> > > 
> > > 
> > > 我现在认为问题源于 Magic Mount，在使用它的 Magisk 和 KernelSU 中，LSPosed 通常被推荐或检测为已安装，并且在启用排除的情况下，MemoryDetector 不会检测到它
> > 
> > 
> > 理解。您可以尝试 overlays 模式，看看是否有任何更改。
> 
> 在 Ovrlayfs 模式下没有发现问题 在 Magic Mount 模式下启用 APatch 的排除模块解决了这个问题 我希望你能告诉更多的人这个 😋

VisionR1, 2025-02-15:
Yeah, sorry i forgot about this.
Need exclusion, and possibly zygisk assistant.

Ruieaas, 2025-02-15:
> 是的，对不起，我忘记了这个（我以为你在排除列表中有它）。需要排除，可能还有 zygisk 助手。

> 是的，对不起，我忘记了这个（我以为你在排除列表中有它）。需要排除，可能还有 zygisk 助手。

Zygisk's adherence to the exclusion list is detected in some detection software and should be disabled in the normal state.

Ruieaas, 2025-02-16:
> 如果你指的是 zygisk 助手，是的，其他人仍然可以检测到。不幸的是，所有这些模块都无法完全隐藏。

APatch may be able to use some hidden modules with Magic Mount support, as well as KPM.

closed 2025-03-09 by Admirepowered