bmax121/APatch #932: Can't use newest apatch on Xiaomi 10 Pro

strayedstar, 2025-02-16:
### Please check before submitting an issue/在提交 issue 前请检查

- [x] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [x] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [x] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [x] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

在崩溃以后断开与apd的连接,导致应用无法管理.

### Reproduce method/复现方法

官核安装10933版本以上的apatch,然后打开超级用户管理,此时崩溃,再次进入丢失秘钥状态并无法恢复,除非重启

### Expected behavior/预期行为

不再崩溃.

### Actual behaviour /实际行为

apatch在故障以后无法校验秘钥并且连接到后端

### Logs/日志

later

### Device Name/设备名称

Xiaomi 10 Pro

### OS Version/系统版本

offical newest hypros

### APatch Version/APatch 版本

above 10933

### Kernel Version/内核版本

4.19

### KernelPatch Version/KernelPatch 版本

0.11

### Additional context/其他信息

[APatch_bugreport_2025-02-16_08_18.tar.gz](https://github.com/user-attachments/files/18813076/APatch_bugreport_2025-02-16_08_18.tar.gz)

Ruieaas, 2025-02-16:
> ### Please check before submit an issue/在提交 issue 前请检查
> * [x] I have searches the issues and have not find any relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容[x] 如果修补失败、root 失败或设备在刷写新的 boot.img 后无法启动。Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问[x] 我会在 APatch Manager - 设置 - 报告日志中上传 bugreport 文件/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志[x] 我知道如何重现可能不特定于我的设备的问题/我知道如何重新复现这个问题
> 
> ### Version requirement/版本要求
> * [x] 我正在使用 APatch/的最新 CI 版本
> 
> ### describe the bug/描述 bug
> 在崩溃以后断开与apd的连接，导致应用无法管理.
> 
> ### Reproduce method/复现方法
> 官核安装10933版本以上的apatch，然后打开超级用户管理，此时崩溃，再次进入丢失秘钥状态并无法恢复，除非重启
> 
> ### Expected behavior/预期行为
> 不再崩溃.
> 
> ### Actual behaviour / 实际行为
> apatch在故障以后无法校验秘钥并且连接到后端
> 
> ### 截图/screenshots
> _无响应_
> 
> ### 日志/日志
> 后
> 
> ### Device Name/设备名称
> 小米 10 Pro
> 
> ### OS Version/系统版本
> 官方最新的 HYPROS
> 
> ### APatch 版本/APatch 版本
> 10933 以上
> 
> ### Kernel Version/内核版本
> 4.19
> 
> ### KernelPatch Version/KernelPatch 版本
> 0.11
> 
> ### Additional context/其他信息
> [APatch_bugreport_2025-02-16_08_18.tar.gz](https://github.com/user-attachments/files/18813076/APatch_bugreport_2025-02-16_08_18.tar.gz)

Try to use the latest APatch, if the problem persists you can upload the logs

strayedstar, 2025-02-16:
> Try to use the latest APatch, if the problem persists you can upload the logs

I use nlightly build,but it still not work.

[APatch_bugreport_2025-02-16_16_58.tar.gz](https://github.com/user-attachments/files/18814171/APatch_bugreport_2025-02-16_16_58.tar.gz)

Ruieaas, 2025-02-17:
> > > ### Please check before submit an issue/在提交 issue 前请检查
> > > 
> > > 
> > > * [x] I have searches the issues and have not find any related/我已经搜索了 issues 列表，没有发现于本问题相关内容[x] 如果修补失败、root 失败或设备在刷写新的 boot.img 后无法启动。Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问[x] 我会在 APatch Manager - 设置 - 报告日志中上传 bugreport 文件/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志[x] 我知道如何重现可能不特定于我的设备的问题/我知道如何重新复现这个问题
> > > ### Version requirement/版本要求
> > > 
> > > 
> > > * [x] 我正在使用 APatch/的最新 CI 版本
> > > ### describe the bug/描述 bug
> > > 在崩溃以后断开与apd的连接，导致应用无法管理.
> > > ### Reproduce method/复现方法
> > > 官核安装10933版本以上的apatch，然后打开超级用户管理，此时崩溃，再次进入丢失秘钥状态并无法恢复，除非重启
> > > ### Expected behavior/预期行为
> > > 不再崩溃.
> > > ### Actual behaviour / 实际行为
> > > apatch在故障以后无法校验秘钥并且连接到后端
> > > ### 截图/screenshots
> > > _无响应_
> > > ### 日志/日志
> > > 后
> > > ### Device Name/设备名称
> > > 小米 10 Pro
> > > ### OS Version/系统版本
> > > 官方最新的 HYPROS
> > > ### APatch 版本/APatch 版本
> > > 10933 以上
> > > ### Kernel Version/内核版本
> > > 4.19
> > > ### KernelPatch Version/KernelPatch 版本
> > > 0.11
> > > ### Additional context/其他信息
> > > [APatch_bugreport_2025-02-16_08_18.tar.gz](https://github.com/user-attachments/files/18813076/APatch_bugreport_2025-02-16_08_18.tar.gz)
> > 
> > 
> > 尝试使用最新的 APatch，如果问题仍然存在，您可以上传日志
> 
> 我使用 nlightly build，但它仍然不起作用。
> 
> [APatch_bugreport_2025-02-16_16_58.tar.gz](https://github.com/user-attachments/files/18814171/APatch_bugreport_2025-02-16_16_58.tar.gz)

KernelSU installed？？？

strayedstar, 2025-02-17:
official kernel not support kernel.
---- 回复的原邮件 ----
发件人 ***@***.***> 日期 2025年02月17日 20:56 收件人 ***@***.***> 抄送至 ***@***.***>***@***.***> 主题 Re: [bmax121/APatch] Can't use newest apatch on Xiaomi 10 Pro (Issue #932) 
Please check before submit an issue/在提交 issue 前请检查
I have searches the issues and have not find any related/我已经搜索了 issues 列表，没有发现于本问题相关内容[x] 如果修补失败、root 失败或设备在刷写新的 boot.img 后无法启动。Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问[x] 我会在 APatch Manager - 设置 - 报告日志中上传 bugreport 文件/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志[x] 我知道如何重现可能不特定于我的设备的问题/我知道如何重新复现这个问题
Version requirement/版本要求
我正在使用 APatch/的最新 CI 版本
describe the bug/描述 bug
在崩溃以后断开与apd的连接，导致应用无法管理.
Reproduce method/复现方法
官核安装10933版本以上的apatch，然后打开超级用户管理，此时崩溃，再次进入丢失秘钥状态并无法恢复，除非重启
Expected behavior/预期行为
不再崩溃.
Actual behaviour / 实际行为
apatch在故障以后无法校验秘钥并且连接到后端
截图/screenshots
无响应
日志/日志
后
Device Name/设备名称
小米 10 Pro
OS Version/系统版本
官方最新的 HYPROS
APatch 版本/APatch 版本
10933 以上
Kernel Version/内核版本
4.19
KernelPatch Version/KernelPatch 版本
0.11
Additional context/其他信息
APatch_bugreport_2025-02-16_08_18.tar.gz
尝试使用最新的 APatch，如果问题仍然存在，您可以上传日志
我使用 nlightly build，但它仍然不起作用。
APatch_bugreport_2025-02-16_16_58.tar.gz
KernelSU installed？？？
—
Please check before submit an issue/在提交 issue 前请检查
I have searches the issues and have not find any related/我已经搜索了 issues 列表，没有发现于本问题相关内容[x] 如果修补失败、root 失败或设备在刷写新的 boot.img 后无法启动。Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问[x] 我会在 APatch Manager - 设置 - 报告日志中上传 bugreport 文件/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志[x] 我知道如何重现可能不特定于我的设备的问题/我知道如何重新复现这个问题
Version requirement/版本要求
我正在使用 APatch/的最新 CI 版本
describe the bug/描述 bug
在崩溃以后断开与apd的连接，导致应用无法管理.
Reproduce method/复现方法
官核安装10933版本以上的apatch，然后打开超级用户管理，此时崩溃，再次进入丢失秘钥状态并无法恢复，除非重启
Expected behavior/预期行为
不再崩溃.
Actual behaviour / 实际行为
apatch在故障以后无法校验秘钥并且连接到后端
截图/screenshots
无响应
日志/日志
后
Device Name/设备名称
小米 10 Pro
OS Version/系统版本
官方最新的 HYPROS
APatch 版本/APatch 版本
10933 以上
Kernel Version/内核版本
4.19
KernelPatch Version/KernelPatch 版本
0.11
Additional context/其他信息
APatch_bugreport_2025-02-16_08_18.tar.gz
尝试使用最新的 APatch，如果问题仍然存在，您可以上传日志
我使用 nlightly build，但它仍然不起作用。
APatch_bugreport_2025-02-16_16_58.tar.gz
KernelSU installed？？？

strayedstar, 2025-02-17:
not support kernelsu.
---- 回复的原邮件 ----
发件人 ***@***.***> 日期 2025年02月17日 20:56 收件人 ***@***.***> 抄送至 ***@***.***>***@***.***> 主题 Re: [bmax121/APatch] Can't use newest apatch on Xiaomi 10 Pro (Issue #932) 
Please check before submit an issue/在提交 issue 前请检查
I have searches the issues and have not find any related/我已经搜索了 issues 列表，没有发现于本问题相关内容[x] 如果修补失败、root 失败或设备在刷写新的 boot.img 后无法启动。Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问[x] 我会在 APatch Manager - 设置 - 报告日志中上传 bugreport 文件/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志[x] 我知道如何重现可能不特定于我的设备的问题/我知道如何重新复现这个问题
Version requirement/版本要求
我正在使用 APatch/的最新 CI 版本
describe the bug/描述 bug
在崩溃以后断开与apd的连接，导致应用无法管理.
Reproduce method/复现方法
官核安装10933版本以上的apatch，然后打开超级用户管理，此时崩溃，再次进入丢失秘钥状态并无法恢复，除非重启
Expected behavior/预期行为
不再崩溃.
Actual behaviour / 实际行为
apatch在故障以后无法校验秘钥并且连接到后端
截图/screenshots
无响应
日志/日志
后
Device Name/设备名称
小米 10 Pro
OS Version/系统版本
官方最新的 HYPROS
APatch 版本/APatch 版本
10933 以上
Kernel Version/内核版本
4.19
KernelPatch Version/KernelPatch 版本
0.11
Additional context/其他信息
APatch_bugreport_2025-02-16_08_18.tar.gz
尝试使用最新的 APatch，如果问题仍然存在，您可以上传日志
我使用 nlightly build，但它仍然不起作用。
APatch_bugreport_2025-02-16_16_58.tar.gz
KernelSU installed？？？
—
Please check before submit an issue/在提交 issue 前请检查
I have searches the issues and have not find any related/我已经搜索了 issues 列表，没有发现于本问题相关内容[x] 如果修补失败、root 失败或设备在刷写新的 boot.img 后无法启动。Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问[x] 我会在 APatch Manager - 设置 - 报告日志中上传 bugreport 文件/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志[x] 我知道如何重现可能不特定于我的设备的问题/我知道如何重新复现这个问题
Version requirement/版本要求
我正在使用 APatch/的最新 CI 版本
describe the bug/描述 bug
在崩溃以后断开与apd的连接，导致应用无法管理.
Reproduce method/复现方法
官核安装10933版本以上的apatch，然后打开超级用户管理，此时崩溃，再次进入丢失秘钥状态并无法恢复，除非重启
Expected behavior/预期行为
不再崩溃.
Actual behaviour / 实际行为
apatch在故障以后无法校验秘钥并且连接到后端
截图/screenshots
无响应
日志/日志
后
Device Name/设备名称
小米 10 Pro
OS Version/系统版本
官方最新的 HYPROS
APatch 版本/APatch 版本
10933 以上
Kernel Version/内核版本
4.19
KernelPatch Version/KernelPatch 版本
0.11
Additional context/其他信息
APatch_bugreport_2025-02-16_08_18.tar.gz
尝试使用最新的 APatch，如果问题仍然存在，您可以上传日志
我使用 nlightly build，但它仍然不起作用。
APatch_bugreport_2025-02-16_16_58.tar.gz
KernelSU installed？？？

Ruieaas, 2025-02-18:
> 不支持 kernelsu。
> ---- 回复的原邮件 ----
> 发件人 ***@***.***> 日期 2025年02月17日 20：56 收件人 ***@***.***> 抄送至 ***@***.***>***@***.***> 主题 Re： [bmax121/APatch] 小米 10 Pro 无法使用最新的补丁 （Issue [#932](https://github.com/bmax121/APatch/issues/932)）
> Please check before submit an issue/在提交 issue 前请检查
> I have searches the issues and have not find any related/我已经搜索了 issues 列表，没有发现于本问题相关内容[x] 如果修补失败、root 失败或设备在刷写新的 boot.img 后无法启动。Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问[x] 我会在 APatch Manager - 设置 - 报告日志中上传 bugreport 文件/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志[x] 我知道如何重现可能不特定于我的设备的问题/我知道如何重新复现这个问题
> Version requirement/版本要求
> 我正在使用 APatch/的最新 CI 版本
> describe the bug/描述 bug
> 在崩溃以后断开与apd的连接，导致应用无法管理.
> Reproduce method/复现方法
> 官核安装10933版本以上的apatch，然后打开超级用户管理，此时崩溃，再次进入丢失秘钥状态并无法恢复，除非重启
> Expected behavior/预期行为
> 不再崩溃.
> Actual behaviour / 实际行为
> apatch在故障以后无法校验秘钥并且连接到后端
> 截图/screenshots
> 无响应
> 日志/日志
> 后
> Device Name/设备名称
> 小米 10 Pro
> OS Version/系统版本
> 官方最新的 HYPROS
> APatch 版本/APatch 版本
> 10933 以上
> Kernel Version/内核版本
> 4.19
> KernelPatch Version/KernelPatch 版本
> 0.11
> Additional context/其他信息
> APatch_bugreport_2025-02-16_08_18.tar.gz
> 尝试使用最新的 APatch，如果问题仍然存在，您可以上传日志
> 我使用 nlightly build，但它仍然不起作用。
> APatch_bugreport_2025-02-16_16_58.tar.gz
> 已安装 KernelSU???
> —
> 直接回复此电子邮件，在 GitHub 上查看，或取消订阅。
> 您收到此消息是因为您编写了该线程。消息 ID：***@***.***>
> Ruieaas 留言 （[bmax121/APatch#932](https://github.com/bmax121/APatch/issues/932)）
> Please check before submit an issue/在提交 issue 前请检查
> I have searches the issues and have not find any related/我已经搜索了 issues 列表，没有发现于本问题相关内容[x] 如果修补失败、root 失败或设备在刷写新的 boot.img 后无法启动。Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问[x] 我会在 APatch Manager - 设置 - 报告日志中上传 bugreport 文件/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志[x] 我知道如何重现可能不特定于我的设备的问题/我知道如何重新复现这个问题
> Version requirement/版本要求
> 我正在使用 APatch/的最新 CI 版本
> describe the bug/描述 bug
> 在崩溃以后断开与apd的连接，导致应用无法管理.
> Reproduce method/复现方法
> 官核安装10933版本以上的apatch，然后打开超级用户管理，此时崩溃，再次进入丢失秘钥状态并无法恢复，除非重启
> Expected behavior/预期行为
> 不再崩溃.
> Actual behaviour / 实际行为
> apatch在故障以后无法校验秘钥并且连接到后端
> 截图/screenshots
> 无响应
> 日志/日志
> 后
> Device Name/设备名称
> 小米 10 Pro
> OS Version/系统版本
> 官方最新的 HYPROS
> APatch 版本/APatch 版本
> 10933 以上
> Kernel Version/内核版本
> 4.19
> KernelPatch Version/KernelPatch 版本
> 0.11
> Additional context/其他信息
> APatch_bugreport_2025-02-16_08_18.tar.gz
> 尝试使用最新的 APatch，如果问题仍然存在，您可以上传日志
> 我使用 nlightly build，但它仍然不起作用。
> APatch_bugreport_2025-02-16_16_58.tar.gz
> 已安装 KernelSU???
> —
> 直接回复此电子邮件，在 GitHub 上查看，或取消订阅。
> 您收到此消息是因为您编写了该线程。消息 ID：***@***.***>

Third party systems？

strayedstar, 2025-02-18:
offical system.
---- 回复的原邮件 ----
发件人 ***@***.***> 日期 2025年02月18日 12:17 收件人 ***@***.***> 抄送至 ***@***.***>***@***.***> 主题 Re: [bmax121/APatch] Can't use newest apatch on Xiaomi 10 Pro (Issue #932) 
不支持 kernelsu。
---- 回复的原邮件 ----
发件人 @.> 日期 2025年02月17日 20：56 收件人 @.> 抄送至 @.>@.> 主题 Re： [bmax121/APatch] 小米 10 Pro 无法使用最新的补丁 （Issue #932）
Please check before submit an issue/在提交 issue 前请检查
I have searches the issues and have not find any related/我已经搜索了 issues 列表，没有发现于本问题相关内容[x] 如果修补失败、root 失败或设备在刷写新的 boot.img 后无法启动。Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问[x] 我会在 APatch Manager - 设置 - 报告日志中上传 bugreport 文件/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志[x] 我知道如何重现可能不特定于我的设备的问题/我知道如何重新复现这个问题
Version requirement/版本要求
我正在使用 APatch/的最新 CI 版本
describe the bug/描述 bug
在崩溃以后断开与apd的连接，导致应用无法管理.
Reproduce method/复现方法
官核安装10933版本以上的apatch，然后打开超级用户管理，此时崩溃，再次进入丢失秘钥状态并无法恢复，除非重启
Expected behavior/预期行为
不再崩溃.
Actual behaviour / 实际行为
apatch在故障以后无法校验秘钥并且连接到后端
截图/screenshots
无响应
日志/日志
后
Device Name/设备名称
小米 10 Pro
OS Version/系统版本
官方最新的 HYPROS
APatch 版本/APatch 版本
10933 以上
Kernel Version/内核版本
4.19
KernelPatch Version/KernelPatch 版本
0.11
Additional context/其他信息
APatch_bugreport_2025-02-16_08_18.tar.gz
尝试使用最新的 APatch，如果问题仍然存在，您可以上传日志
我使用 nlightly build，但它仍然不起作用。
APatch_bugreport_2025-02-16_16_58.tar.gz
已安装 KernelSU???
—
直接回复此电子邮件，在 GitHub 上查看，或取消订阅。
您收到此消息是因为您编写了该线程。消息 ***@***.***>
Ruieaas 留言 （bmax121/APatch#932）
Please check before submit an issue/在提交 issue 前请检查
I have searches the issues and have not find any related/我已经搜索了 issues 列表，没有发现于本问题相关内容[x] 如果修补失败、root 失败或设备在刷写新的 boot.img 后无法启动。Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问[x] 我会在 APatch Manager - 设置 - 报告日志中上传 bugreport 文件/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志[x] 我知道如何重现可能不特定于我的设备的问题/我知道如何重新复现这个问题
Version requirement/版本要求
我正在使用 APatch/的最新 CI 版本
describe the bug/描述 bug
在崩溃以后断开与apd的连接，导致应用无法管理.
Reproduce method/复现方法
官核安装10933版本以上的apatch，然后打开超级用户管理，此时崩溃，再次进入丢失秘钥状态并无法恢复，除非重启
Expected behavior/预期行为
不再崩溃.
Actual behaviour / 实际行为
apatch在故障以后无法校验秘钥并且连接到后端
截图/screenshots
无响应
日志/日志
后
Device Name/设备名称
小米 10 Pro
OS Version/系统版本
官方最新的 HYPROS
APatch 版本/APatch 版本
10933 以上
Kernel Version/内核版本
4.19
KernelPatch Version/KernelPatch 版本
0.11
Additional context/其他信息
APatch_bugreport_2025-02-16_08_18.tar.gz
尝试使用最新的 APatch，如果问题仍然存在，您可以上传日志
我使用 nlightly build，但它仍然不起作用。
APatch_bugreport_2025-02-16_16_58.tar.gz
已安装 KernelSU???
—
直接回复此电子邮件，在 GitHub 上查看，或取消订阅。
您收到此消息是因为您编写了该线程。消息 ***@***.***>
Third party systems？
—
不支持 kernelsu。
---- 回复的原邮件 ----
发件人 @.> 日期 2025年02月17日 20：56 收件人 @.> 抄送至 @.>@.> 主题 Re： [bmax121/APatch] 小米 10 Pro 无法使用最新的补丁 （Issue #932）
Please check before submit an issue/在提交 issue 前请检查
I have searches the issues and have not find any related/我已经搜索了 issues 列表，没有发现于本问题相关内容[x] 如果修补失败、root 失败或设备在刷写新的 boot.img 后无法启动。Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问[x] 我会在 APatch Manager - 设置 - 报告日志中上传 bugreport 文件/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志[x] 我知道如何重现可能不特定于我的设备的问题/我知道如何重新复现这个问题
Version requirement/版本要求
我正在使用 APatch/的最新 CI 版本
describe the bug/描述 bug
在崩溃以后断开与apd的连接，导致应用无法管理.
Reproduce method/复现方法
官核安装10933版本以上的apatch，然后打开超级用户管理，此时崩溃，再次进入丢失秘钥状态并无法恢复，除非重启
Expected behavior/预期行为
不再崩溃.
Actual behaviour / 实际行为
apatch在故障以后无法校验秘钥并且连接到后端
截图/screenshots
无响应
日志/日志
后
Device Name/设备名称
小米 10 Pro
OS Version/系统版本
官方最新的 HYPROS
APatch 版本/APatch 版本
10933 以上
Kernel Version/内核版本
4.19
KernelPatch Version/KernelPatch 版本
0.11
Additional context/其他信息
APatch_bugreport_2025-02-16_08_18.tar.gz
尝试使用最新的 APatch，如果问题仍然存在，您可以上传日志
我使用 nlightly build，但它仍然不起作用。
APatch_bugreport_2025-02-16_16_58.tar.gz
已安装 KernelSU???
直接回复此电子邮件，在 GitHub 上查看，或取消订阅。
您收到此消息是因为您编写了该线程。消息 ***@***.***>
Ruieaas 留言 （bmax121/APatch#932）
Please check before submit an issue/在提交 issue 前请检查
I have searches the issues and have not find any related/我已经搜索了 issues 列表，没有发现于本问题相关内容[x] 如果修补失败、root 失败或设备在刷写新的 boot.img 后无法启动。Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问[x] 我会在 APatch Manager - 设置 - 报告日志中上传 bugreport 文件/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志[x] 我知道如何重现可能不特定于我的设备的问题/我知道如何重新复现这个问题
Version requirement/版本要求
我正在使用 APatch/的最新 CI 版本
describe the bug/描述 bug
在崩溃以后断开与apd的连接，导致应用无法管理.
Reproduce method/复现方法
官核安装10933版本以上的apatch，然后打开超级用户管理，此时崩溃，再次进入丢失秘钥状态并无法恢复，除非重启
Expected behavior/预期行为
不再崩溃.
Actual behaviour / 实际行为
apatch在故障以后无法校验秘钥并且连接到后端
截图/screenshots
无响应
日志/日志
后
Device Name/设备名称
小米 10 Pro
OS Version/系统版本
官方最新的 HYPROS
APatch 版本/APatch 版本
10933 以上
Kernel Version/内核版本
4.19
KernelPatch Version/KernelPatch 版本
0.11
Additional context/其他信息
APatch_bugreport_2025-02-16_08_18.tar.gz
尝试使用最新的 APatch，如果问题仍然存在，您可以上传日志
我使用 nlightly build，但它仍然不起作用。
APatch_bugreport_2025-02-16_16_58.tar.gz
已安装 KernelSU???
直接回复此电子邮件，在 GitHub 上查看，或取消订阅。
您收到此消息是因为您编写了该线程。消息 ***@***.***>
Third party systems？

marat2509, 2025-02-18:
@strayedstar @Ruieaas - stop overquoting messages