bmax121/APatch #934: [Xperia 5 V] Touchscreen No Response After Patching

alex88510, 2025-02-16:
### Please check before submitting an issue/在提交 issue 前请检查

- [x] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [x] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [x] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [x] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

The patching of the boot.img is successful. 
The flashing of the patched boot.img is also successful.
The device is also able to boot after flashing the patched boot.img. 

However, the touchscreen has no response, couldn't interact with the screen at all. 
The power button seems to be responding (including the fingerprint sensor) but without the touchscreen, the device is still not usable. 

### Reproduce method/复现方法

1. Patch boot.img
2. Flash boot.img
3. Reboot

### Expected behavior/预期行为

Touchscreen should be functioning as usual. 

### Actual behaviour /实际行为

Touchscreen not working

### Logs/日志

This is the log when patching the boot.img

`****************************
****************************
 APatch Boot Image Patcher
****************************
- Patching kernel
+ ./kptools -p -i kernel.ori -S _<redacted>_ -k kpimg -o kernel
[+] kernel image_size: 0x03087a00
[+] kernel uefi header: true
[+] kernel load_offset: 0x00000000
[+] kernel kernel_size: 0x03210000
[+] kernel page_shift: 12
[+] new kernel image ...
[+] linux_banner 1: Linux version 5.15.175-gb2b46429e6ce (root@61157ecdef21) (Android (12328485, +pgo, +bolt, +lto, +mlgo, based on r530567) clang version 19.0.0 (https://android.googlesource.com/toolchain/llvm-project 97a699bf4812a18fb657c2779f5296a4ab2694d2), LLD 19.0.0) #1 SMP PREEMPT Wed Feb 12 10:50:47 UTC 2025
[+] linux_banner offset: 0x2126010
[+] kernel version major: 5, minor: 15, patch: 175
[+] kallsyms_token_table offset: 0x01e7c710
[+] endian: little
[+] kallsyms_token_index offset: 0x01e7cad0
[+] arm64 relocation kernel_va: 0xffffffffffffffff
[?] can't find arm64 relocation table
[+] kallsyms_markers range: [0x01e7bc10, 0x01e7c70c), count: 0x000002be
[+] approximate kallsyms_offsets range: [0x01b3e960, 0x01bee494) count: 0x0002becd
[+] kallsyms_names offset: 0x01bee4a0
[?] can't find kallsyms_num_syms, try: 0x0002bec3
[+] names table linux_banner index: 0x00019264
[+] linux_banner index: 0
[+] kallsyms_offsets offset: 0x01b3e960
[+] pid_vnr: type: T, offset: 0x00190730
[+] pid_vnr verfied sp_el0, insn: 0xd5384108
[+] layout kimg: 0x0,0x3087a00, kpimg: 0x3088000,0x2c110, extra: 0x30b4110,0x80, end: 0x30b4190, start: 0x3210000
[+] kpimg version: b02
[+] kpimg compile time: 01:05:21 Feb 12 2025
[+] kpimg config: android, release
[+] tcp_init_sock: type: T, offset: 0x013f4ab0
[+] map_start: 0x13f4ab0, max_size: 0x800
[+] kallsyms_lookup_name: type: T, offset: 0x002af4c4
[?] no symbol: printk
[+] _printk: type: T, offset: 0x00212960
[+] memblock_reserve: type: T, offset: 0x004ccc10
[+] memblock_free: type: T, offset: 0x004cb528
[+] memblock_mark_nomap: type: T, offset: 0x004ccf20
[+] memblock_phys_alloc_try_nid: type: T, offset: 0x028d6e88
[?] no symbol: memblock_virt_alloc_try_nid
[+] memblock_alloc_try_nid: type: T, offset: 0x028d716c
[+] panic: type: T, offset: 0x0014ca84
[+] rest_init: type: T, offset: 0x016ee7c8
[+] kernel_init: type: t, offset: 0x016ee8b4
[?] no symbol: report_cfi_failure
[+] __cfi_slowpath_diag: type: T, offset: 0x003dde24
[?] no symbol: __cfi_slowpath
[+] copy_process: type: t, offset: 0x00145620
[+] avc_denied: type: t, offset: 0x0088c1e8
[+] slow_avc_audit: type: T, offset: 0x0088a5cc
[+] input_handle_event: type: t, offset: 0x00ef6aac
[+] root superkey hash: _<redacted>_
[+] paging_init: type: T, offset: 0x028c1cf4
[+] patch done: kernel
+ patch_rc=0
+ set +x
- Repacking boot image
- Successfully Patched!
 Output file is written to 
 /storage/emulated/0/Download/apatch_patched_11039_0.11.2_vhoe.img
****************************
`

### Device Name/设备名称

Sony Xperia 5 V

### OS Version/系统版本

Lineage OS 22.1 (Android 15)

### APatch Version/APatch 版本

11039

### Kernel Version/内核版本

5.15.175

### KernelPatch Version/KernelPatch 版本

0.11.2

### Additional context/其他信息

Since I cannot unlock the phone, I couldn't get the log from the APatch Manager when the issue occurs.

pedroh77, 2025-02-17:
Happening to me as well. Differences: device is Moto g9 Power, kernel is 4.19.325-perf+. Apatch is latest release, not CI. If it matters, this patched boot image has Cherish Peekaboo 1.5 embedded.

Of note, I am also trying to migrate back from Apatch Next. It's running right now, with the same Superkey too. Can this break the touchscreen somehow? I would have thought that Next would just stop working when I boot with a different patched boot image.

Other than that, it's all the same.

VisionR1, 2025-02-18:
About the logs, have some in /data/adb/ap/log, with a recovery and a pc, can see it.

pedroh77, 2025-02-18:
I have since solved it. For me at least, the probable culprit was Apatch Next being still there, and maybe also that I used the same superkey to patch the bootloader with the other Apatch.

l don't remember what did it: whether it was changing the superkey (maybe unnecessary, if you can make Apatch / its fork forget it), or just getting rid of the coexisting Apatch fork altogether.

So all in all, though this may or may not be related to this report, since Apatch Next just recently announced an update, and it turns out it's regular Apatch, and also Apatch Next is discontinued, for all those in the same situation as me I would recommend:

1) Use a different superkey with Apatch or, if that and if other middle ground attempts fail;
2) Make Apatch Next unroot the device, which includes patching back the boot image, reboot, uninstall Next, and proceed as normal with Apatch;
3) Or just stay on Apatch Next until stuff breaks. Always a reasonable choice if things are working for you right now;

Also in my device Cherish Peekaboo broke current Apatch (release and CI) every time no matter its version, but then I digress, as the symptom is not losing control of touchscreen so, just something I wanted to get out there. Just keep in mind that if it's not working, try again without embedding it.

Kaukov, 2025-02-27:
I had this happen to me on an Xperia 1V. I had Magisk before, but I reverted back to the original `boot.img` and `init_boot.img` before trying APatch.

The power and volume buttons also refuse to work. I tried flashing the stock `boot.img` again, but I rebooted into recovery and I still can't do anything, even reboot. So now the phone is stuck in recovery, the screen turned off and I can't turn it back on. I'll wait for the battery to drain by itself and hope I can enter bootloader mode to re-flash the partitions again.

ewigl, 2025-02-27:
> I had this happen to me on an Xperia 1V with LineageOS 22.1.

Same device and ROM here.

 **The touchscreen doesn’t work**, so I unlocked my phone using scrcpy and found that **Apatch works fine**.

I have another phone running **Apatch** with **LineageOS 22.1**, and it works without any issues.

I will try it later.

znshje, 2025-02-27:
Same issue on my Xperia 1V, crDroid 11.1 beta. I have to flash back the original `boot.img` for my phone to work well.

Gottoz, 2025-03-02:
The same on oneplus11 linageos 22 boot log 

[log.txt](https://github.com/user-attachments/files/19042481/log.txt)

mareliada, 2025-03-13:
I have also the exact same issue occurring on my XQ-DQ54 with LOS 22.1.

jjbr, 2025-07-23:
Likewise, 11107 seems to have fixed the issues with LOS 22.1 on my Xperia V 5 (patched boot image, flashed via fastboot) 🎊