bmax121/APatch #942: 更新至11039后模块安装失败问题

geekloong, 2025-02-20:
### Please check before submitting an issue/在提交 issue 前请检查

- [x] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [x] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [x] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [x] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

我从旧版本更新到11039后发现模块无论是卸载也好，还是安装也好，都无法正常生效，甚至卸载之后它仍然存在。

### Reproduce method/复现方法

复现：正常下载最新版APP，直接覆盖安装，然后修补镜像，fastboot刷入，重启，发现模块掉了，刷入新模块，模块不生效。

### Expected behavior/预期行为

模块应该生效

### Actual behaviour /实际行为

模块删不掉且打开时不生效。

### Screenshots/截图

[image]

### Device Name/设备名称

Redmi K50U

### OS Version/系统版本

android14

### APatch Version/APatch 版本

11039

### Kernel Version/内核版本

5.10.x

### KernelPatch Version/KernelPatch 版本

0.11.2

### Additional context/其他信息

日志太大，不便上传

geekloong, 2025-02-20:
临时解决方法，老版本与新完毕不兼容问题，使用下面步骤可恢复正常：
1.删除data/adb下除了【ap】文件夹和【apd】文件以外的任何内容。
2.打开apatch点击【完全卸载】。
3.重新修补镜像并刷写。
希望能帮到同等疑惑的朋友们，希望后续此项目的开发者们能写一些代码删除掉这些不兼容的问题。

VisionR1, 2025-02-20:
I think that you mentioned, is relative with this.

[image]

306bobby-android, 2025-02-20:
> I think that you mentioned, is relative with this.
> 
> [image]

They mentioned that after disappearing he reinstalled it and it still didn't show up

VisionR1, 2025-02-20:
> > I think that you mentioned, is relative with this.
> > [image]
> 
> They mentioned that after disappearing he reinstalled it and it still didn't show up

So, first delete all in /data/adb, and if not working, clean install.

geekloong, 2025-02-21:
> > > I think that you mentioned, is relative with this.我认为你提到的，与此是相关的。
> > > [image]
> > 
> > 
> > They mentioned that after disappearing he reinstalled it and it still didn't show up他们提到，消失后他重新安装了它，但它仍然没有出现
> 
> So, first delete all in /data/adb, and if not working, clean install.因此，首先删除 /data/adb 中的所有内容，如果不起作用，则进行全新安装。

后续我清理进行了全新安装，仅生效了一次，我昨晚重启了手机，接着所有模块就又失效了，也出现了卸载不掉的情况。
问题仍然没有解决。

[image]

VisionR1, 2025-02-21:
Strange, so you complete uninstall and patch the boot and clean install and still the same, this mean something about your device interactive and causing this issue.

changliuxian, 2025-02-23:
同样的问题，照这方法完全卸载后重新修补重启lsp又消失了。回旧版去了。

zyklone4096, 2025-02-27:
相同问题，OnePlus ACE3

zyklone4096, 2025-02-27:
找到一个很邪门的解决方法，临时切换到Magisk安装LSPosed，然后恢复init_boot刷入apatch的boot，但也有可能是偶发问题

Admirepowered, 2025-03-09:
找到个可能的问题，大概率不是ap问题，救砖模块不要装试试

commit 73b5cfd1 (2025-05-11): fix #942
commit 5a1cf9cd (2025-05-11): fix #942
closed 2025-05-11 by Admirepowered