bmax121/APatch #944: 已安装zygisk next安装lsposed时遇到问题

yangyyyy15, 2025-02-21:
### Please check before submitting an issue/在提交 issue 前请检查

- [x] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [x] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [x] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [x] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

zygisk next和lsposed都能刷入，zygisk next安装后frida正常，但是lsposed安装后，下拉通知栏，希望进入lsposed，创建快捷方式，会报错

### Reproduce method/复现方法

卸载了zygisk next和lsposed重装数次都这样，重装前确保/data/adb相关的moudles和modules_update内容清空.但重装重启后试图进入lsposed就会失败，不会卡死但会报错.尝试过最新版本的lspossed 1.9.2 （7024）（7058）和更老的1.8.3-6552

### Expected behavior/预期行为

通知栏点击正常进入lsposed

### Actual behaviour /实际行为

system没有响应/shell屡次停止运行

### Screenshots/截图

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Logs/日志

<html>
<body>
<form name="codeSnippet" id="codeSnippet" class="container" style="box-sizing: border-box; display: flex; height: calc(var(--vh, 1vh) * 100); padding: 0px; margin: 0px; flex-direction: row-reverse; color: rgb(0, 0, 0); font-family: &quot;Fira Code v2&quot;, monospace; font-size: medium; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><div id="main" class="content" style="box-sizing: border-box; flex: 10 1 0%; max-height: 100%; max-width: 100%; overflow-wrap: break-word;"><code name="snippet" spellcheck="false" data-gramm="false" readonly="true" id="snippet" class="textarea" style="box-sizing: border-box; margin: 0px; font-family: &quot;Fira Code v2&quot;, monospace; font-size: 10pt; display: inline-block; outline: transparent solid 0px; background: rgb(0, 0, 0); height: 742px; width: 1520.8px; padding: 3em; overflow: auto; color: rgb(238, 238, 238); border: none; resize: none; tab-size: 4; white-space: pre-wrap; overflow-wrap: normal; word-break: normal;">
  | time: 1740101333079
-- | --
  | msg: Native crash: Segmentation fault
  | stacktrace: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
  | PixelOS Version: 'PixelOS_xaga-14.0-20240522-1147'
  | Build fingerprint: 'Redmi/xagapro/xagapro:14/AP1A.240505.005.B1/eng.aadmin.20240522.144720:user/release-keys'
  | Revision: '0'
  | ABI: 'arm64'
  | Timestamp: 2025-02-21 09:28:53.017152311+0800
  | Process uptime: 0s
  | Cmdline: zygote64
  | pid: 5737, tid: 5737, name: m.android.shell  >>> zygote64 <<<
  | uid: 2000
  | tagged_addr_ctrl: 0000000000000001 (PR_TAGGED_ADDR_ENABLE)
  | signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0000000000000000
  | Cause: null pointer dereference
  | x0  0000000000000000  x1  0000000000000000  x2  000000005c000000  x3  0000007fc4f88000
  | x4  0000000000000000  x5  00000072e81c6985  x6  000000000000004a  x7  7f7f7f7f7f7f7f7f
  | x8  0000000000000000  x9  e7b25a2fa1d228f6  x10 0000000000000000  x11 00000072ea200030
  | x12 000000739a742670  x13 000000000000000a  x14 0000000000000048  x15 000000000000000b
  | x16 00000072bd9f4398  x17 000000737a957fc0  x18 000000739af8c000  x19 0000000000000000
  | x20 0000007fc4f886e0  x21 0000000040300002  x22 0000000000000000  x23 b4000072ea25ee00
  | x24 00000072bd9fd000  x25 0000007fc4f88eec  x26 000000720000d7e0  x27 000000739a4dd051
  | x28 00000072bd9f6288  x29 0000000000000011
  | lr  00000072bd9d173c  sp  0000007fc4f88630  pc  000000737a957fd0  pst 0000000080001000
  | 2 total frames
  | backtrace:
[... 69 lines cut ...]
  | ABI: 'arm64'
  | Timestamp: 2025-02-21 10:04:39.354787387+0800
  | Process uptime: 0s
  | Cmdline: zygote64
  | pid: 5576, tid: 5576, name: m.android.shell  >>> zygote64 <<<
  | uid: 2000
  | tagged_addr_ctrl: 0000000000000001 (PR_TAGGED_ADDR_ENABLE)
  | signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0000000000000000
  | Cause: null pointer dereference
  | x0  0000000000000000  x1  0000000000000000  x2  000000005c000000  x3  0000007fcd7a2e20
  | x4  0000000000000000  x5  0000006fcc505985  x6  000000000000004a  x7  7f7f7f7f7f7f7f7f
  | x8  0000000000000000  x9  6541eb2bfa92caae  x10 0000000000000000  x11 0000006fd1000030
  | x12 00000070844a0670  x13 000000000000000a  x14 0000000000000048  x15 000000000000000b
  | x16 0000006fa2b72290  x17 000000707a0f3fc0  x18 00000070855f0000  x19 0000000000000000
  | x20 0000007fcd7a3500  x21 0000000040300002  x22 0000000000000000  x23 b400006fd105ee00
  | x24 0000006fa2b7b000  x25 0000007fcd7a3d0c  x26 0000006ec000d840  x27 000000708423b051
  | x28 0000006fa2b73b08  x29 0000000000000011
  | lr  0000006fa2b4ece0  sp  0000007fcd7a3450  pc  000000707a0f3fd0  pst 0000000080001000
  | 2 total frames
  | backtrace:
  | #00 pc 00000000000acfd0  /apex/com.android.runtime/lib64/bionic/libc.so (__strlen_aarch64+16) (BuildId: 471cedc15a65b1f48903c1c722513647)
  | #01 pc 000000000000bcdc  /memfd:jit-cache-zygisk_lsposed (deleted)

</body>
</html>

Admirepowered, 2025-02-21:
LSP的问题，不支持Android14 Q2以上的系统，请自行寻找解决方案

VisionR1, 2025-02-21:
Have you try this?
https://github.com/JingMatrix/LSPosed

yangyyyy15, 2025-02-21:
> Have you try this? https://github.com/JingMatrix/LSPosed

not yet. I'm going to try it out right now. thank you for reminding me of this!

yangyyyy15, 2025-02-21:
> LSP的问题，不支持Android14 Q2以上的系统，请自行寻找解决方案

我换了一个lsposed成功了，谢谢你。打扰了

VisionR1, 2025-02-21:
That's is good. 👍🏻

popstarmc, 2025-08-29:
请问解决了吗