bmax121/APatch #958: Substratum Support

Kevinco1, 2025-03-02:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

Please add support for Substratum Lite. Overlays are never applied properly after reboot with AOSP 13-15 and OneUI 6-7

https://play.google.com/store/apps/details?id=projekt.substratum.lite

### Describe the solution you'd like/描述你想要的解决方案

Full overlay support

### Describe alternatives you've considered/描述您考虑过的备选方案

Substratum Lite is fully compatible with Magisk, Alpha, Kitsune and KSU but not APatch

VisionR1, 2025-03-02:
With APatch, you can use magic mount.

VisionR1, 2025-03-03:
> > With APatch, you can use magic mount.
> 
> I'm currently using OneUI 6 - it's possible to mount the root system without any modules. However, I have to mount after each reboot.
> 
> I've installed the overlays...from there you need to reboot and then enable to finalize the theme. This last step is where the problem arises, is this because the root system isn't automatically mounting?

Understand.
Have you try this? 
https://github.com/HuskyDG/magic_overlayfs

306bobby-android, 2025-03-03:
> Understand. Have you try this? https://github.com/HuskyDG/magic_overlayfs

This likely won't help, as this is tied to the same issue with Revanced not mounting at reboot

On both, even running aPatch in OverlayFS mode still breaks things.

Not to mention if the magic mount broke it, it would also be broke on magisk

VisionR1, 2025-03-03:
Yeah, for the latest APatch, not working good the Revanced as i read in TG.
What version of APatch you have?

VisionR1, 2025-03-03:
> > Yeah, for the latest APatch, not working good the Revanced as i read in TG. What version of APatch you have?
> 
> I have reverted to 10763 due to this

Understand.
Have you write in TG about this? Maybe there someone have found a solution.

Kevinco1, 2025-03-03:
> Understand. Have you try this? https://github.com/HuskyDG/magic_overlayfs

I used Husky's overlayfs a while back but it's been archived for over a year now

There are a few forks that are current but I've had trouble getting any to work, they are drastically different with implementation it seems

Kevinco1, 2025-03-03:
These are the ones I'm referring to -

[](https://github.com/backslashxx/magic_overlayfs)
[](https://github.com/Zenlua/Overlayfs)
[](https://github.com/idyll5988/APatch-KSU-Partition-Read-Write)
[](https://github.com/agreenbhm/magic_overlayfs)
[](https://github.com/backslashxx/mountify)

VisionR1, 2025-03-03:
Yeah, you can try it.
I know the last.
https://github.com/backslashxx/mountify
And is update.

Kevinco1, 2025-03-04:
> This likely won't help, as this is tied to the same issue with Revanced not mounting at reboot
>
> On both, even running aPatch in OverlayFS mode still breaks things.
>
> Not to mention if the magic mount [...]

Are you referring to ReVanced Manager because the normal modules of YouTube and Music work just fine with APatch

VisionR1, 2025-03-04:
Really, i read for that.
Because never had use this specific module.

Kevinco1, 2025-03-04:
> Really, i read for that. Because never had use this specific module.

For me, I'd rather use the modules instead of having to patch on my own through the Manager each time there is a new version

VisionR1, 2025-03-04:
> > Really, i read for that. Because never had use this specific module.
> 
> For me, I'd rather use the modules instead of having to patch on my own through the Manager each time there is a new version

Yeah understand, more easy.

306bobby-android, 2025-03-05:
> Are you referring to ReVanced Manager because the normal modules of YouTube and Music work just fine with APatch

I am referencing the revanced manager mount functions.

I use revanced for more than yt/yt music

Kevinco1, 2025-03-05:
> I am referencing the revanced manager mount functions.
>
> I use revanced for more than yt/yt music

What other apps? From what I remember there wasn't many other apps to patch but then again I know there's a few different forks of the manager

306bobby-android, 2025-03-05:
> > > > > > > With APatch, you can use magic mount.
> > > > > > 
> > > > > > 
> > > > > > I'm currently using OneUI 6 - it's possible to mount the root system without any modules. However, I have to mount after each reboot.
> > > > > > I've installed the overlays...from there you need to reboot and then enable to finalize the theme. This last step is where the problem arises, is this because the root system isn't automatically mounting?
> > > > > 
> > > > > 
> > > > > Understand. Have you try this? https://github.com/HuskyDG/magic_overlayfs
> > > > 
> > > > 
> > > > This likely won't help, as this is tied to the same issue with Revanced not mounting at reboot
> > > > On both, even running aPatch in OverlayFS mode still breaks things.
> > > > Not to mention if the magic mount broke it, it would also be broke on magisk
> > > 
> > > 
> > > Are you referring to ReVanced Manager because the normal modules of YouTube and Music work just fine with APatch
> > 
> > 
> > I am referencing the revanced manager mount functions.
> > I use revanced for more than yt/yt music
> 
> What other ones? From what I remember there wasn't many other apps to patch but then again I know there's a few different forks of the manager

The official manager handles many apps and has universal patch sets.

I patch Instagram, yt, yt music, my reddit client, and tiktok when I used it (don't really anymore)

Kevinco1, 2025-03-05:
> The official manager handles many apps and has universal patch sets.
>
> I patch Instagram, yt, yt music, my reddit client, and tiktok when I used it (don't really anymore)

Cool, I'll take another look at it again. And yeah...got rid of TikTok long ago

Kevinco1, 2025-04-10:
Any update on Substratum Lite support being added to APatch eventually?

Mah0Maker, 2025-05-10:
> Any update on Substratum Lite support being added to APatch eventually?

I desperately need that too

Kevinco1, 2025-05-16:
> > Any update on Substratum Lite support being added to APatch eventually?
> 
> I desperately need that too

I'm thinking this is more a Substratum issue. While there has been updates to the main theming app within the past year, there's never been any confirmed APatch or KernelSU support

Mah0Maker, 2025-09-09:
Anyone got it to work?

Kevinco1, 2025-09-10:
> Anyone got it to work?

Unfortunately not, I've tried it with every new action release to no avail