bmax121/APatch #971: 能否出一个只使用内核模块的版本.apatch su会漏环境环境/Can you come up with a version that only uses kernel modules? Apatch su may leak the environment

258967677667, 2025-03-09:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

能否出一个只使用内核模块的版本.apatch su会漏环境环境/Can you come up with a version that only uses kernel modules? Apatch su may leak the environment ,能嵌入内核模块和加载内核模块,加载内核模块,但是没有su/Can embed and load kernel modules, load kernel modules, but without su

### Describe the solution you'd like/描述你想要的解决方案

AP内核模块实在是太好了,但是su会有漏管径的缺陷,请求出一个单独只使用内核模块的/The AP kernel module is really great, but su may have a defect of leaking pipe diameter. Please request a separate one that only uses the kernel module

### Describe alternatives you've considered/描述您考虑过的备选方案

求求了,有一个墓碑冻结模块非常的好用/Please, there is a tombstone freezing module that is very useful

pomelohan, 2025-03-12:
不安装系统补丁，只使用内核补丁即可

closed 2025-03-12 by pomelohan

258967677667, 2025-03-13:
> 不安装系统补丁，只使用内核补丁即可

.