bmax121/APatch #975: Random Restarts and Loss of Superuser Configuration in Latest Version

tofuliang, 2025-03-11:
### Please check before submitting an issue/在提交 issue 前请检查

- [x] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [x] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [ ] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [x] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Random Restarts and Loss of Superuser Configuration after restart

### Reproduce method/复现方法

just wait for it

### Expected behavior/预期行为

Won't auto restarts

### Actual behaviour /实际行为

Random Restarts

### Screenshots/截图

[image]

### Logs/日志

[APatch_bugreport_2025-03-11_11_43.tar.gz](https://github.com/user-attachments/files/19177751/APatch_bugreport_2025-03-11_11_43.tar.gz)

### Device Name/设备名称

Redmi K20 Pro

### OS Version/系统版本

Android 10

### APatch Version/APatch 版本

11039

### Kernel Version/内核版本

4.14.117

### KernelPatch Version/KernelPatch 版本

0.11.2

### Additional context/其他信息

already uploaded

BenceSzalai, 2025-03-15:
I have the same issue:
- Apps selected for Superuser are forgotten time-to-time
- Apps selected for Exclude Modifications are forgotten time-to-time (not related to the above, so sometimes both are forgotten, sometimes only one of the setting types)
- Phone randomly restarts, usually several times in a row, and it is good for a few days after

[image]

Let me know if I can gather some info for further troubleshooting!

tofuliang, 2025-04-03:
It seems that every time there's an app update, the Superuser Configuration gets lost.

KoljaFrahm, 2025-04-08:
If I don't have any modules installed (and reset all settings), I don't get random restarts, just the loss of superuser configuration. If I don't enable any root permission, I guess nothing gets lost (nothing is there to lose), but then I can't use root ^^

hayvan96, 2025-05-08:
Hi, new apatch user. Is there a way to avoid these reboots, which seem to appear when a new app is installed as a matter or fact? I'd like to keep modules (for PIF), but root is not neccessary

KoljaFrahm, 2025-05-14:
I would also want to know

tofuliang, 2025-11-26:
[kernel](https://github.com/Official-Ayrton990/android_kernel_xiaomi_sm8150/releases/download/201220/Quantic-Raphael-201220.zip) from https://github.com/Official-Ayrton990/android_kernel_xiaomi_sm8150/releases/tag/201220
[pstore.zip](https://github.com/user-attachments/files/23759714/pstore.zip) after random reboot

[image]