bmax121/APatch #992: com.android.shell is automatically granted root access on reboot

yezhiyi9670, 2025-03-25:
### Please check before submitting an issue/在提交 issue 前请检查

- [x] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [x] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [x] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [x] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [x] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Even if turned off in the APatch manager, `com.android.shell` is re-granted root access on reboot. This may pose a security risk.

### Reproduce method/复现方法

1. Open APatch manager and turn off root access for `com.android.shell`.
2. Reboot device.
3. Open APatch manager again.
   ❎ Root access for `com.android.shell` is turned on automatically again.

### Expected behavior/预期行为

Root access for `com.android.shell` should remain off.

### Actual behaviour /实际行为

Root access for `com.android.shell` is turned on automatically again.

### Device Name/设备名称

OnePlus 9R (LE2100)

### OS Version/系统版本

ColorOS 14 (LE2100_14.0.0.604)

### APatch Version/APatch 版本

11059

### Kernel Version/内核版本

4.19.157-perf+

### KernelPatch Version/KernelPatch 版本

0.11.3

### Additional context/其他信息

[APatch_bugreport_2025-03-25_11_24.tar.gz](https://github.com/user-attachments/files/19441231/APatch_bugreport_2025-03-25_11_24.tar.gz)

VisionR1, 2025-03-25:
Yeah, is like this by default.

Morrigan-Ship, 2025-04-02:
its because of magisk.
magisk enabling root for shell by default