bmax121/APatch #999: susfs support

n0201, 2025-03-28:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

somewhat. magic mount always gets detected, but i need all the modules i got. I even enabled overlayfs in the newest version by creating the .overlay_enable file, but it didn't help.

### Describe the solution you'd like/描述你想要的解决方案

I'd love working root hiding method which is only possible with susfs at the moment afaik.

### Describe alternatives you've considered/描述您考虑过的备选方案

better magic mount hiding (even after enabling overlayfs magic mount gets detected lol)
Oh and overlayfs also gets detected. seems like both are running at the same time haha

### Additional context/其他信息

i got a few apps that actually search for a lot of methods one of which is magic mount. That's the reason why my banking apps and also other apps won't seem to work. they just refuse to start and some other apps also kinda restrict some functions due to found root.

I'm using: Trickystore, zygisk-next, scene, zygisk assistant, pif (v2-inject), and ios 17.4 emoji font.

and yes, i excluded those apps from modifications in the superuser tab, but it doesn't change anything.

Magisk is even worse and kernelsu next + susfs gki kernels don't want to boot.

YOU are my last chance for a sane clean root setup that is so good, that you can't detect it.

VisionR1, 2025-03-28:
Overlay detect because have ios 17.4 emoji font.

ossilator, 2025-03-29:
see also https://gitlab.com/simonpunk/susfs4ksu/-/issues/38

n0201, 2025-03-29:
Thank you very much for clarification. I kinda got my banking app running. I don't even know how as a reboot (which I already did before ) made them start and not detect root anymore

closed 2025-03-29 by n0201