bmax121/APatch #1001: Apatch hide features

ratankumardas, 2025-03-31:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

Apatch hide.
Implement this feature like this below image.

[image]

### Describe the solution you'd like/描述你想要的解决方案

Apatch hide.
Implement this feature like this below image.

[image]

### Describe alternatives you've considered/描述您考虑过的备选方案

Apatch hide.
Implement this feature like this below image.

[image]

### Additional context/其他信息

Apatch hide.
Implement this feature like this below image.

[image]

VisionR1, 2025-03-31:
It had the same feature, but it was removed because it was causing problems.

ratankumardas, 2025-03-31:
> It had the same feature, but it was removed because it was causing problems.
> 
> [#836](https://github.com/bmax121/APatch/issues/836)
> 
> [#544](https://github.com/bmax121/APatch/issues/544)

I am trying to rename the package and app name but it cannot open. Can you tell me any solution?

VisionR1, 2025-04-01:
> > It had the same feature, but it was removed because it was causing problems.
> > [#836](https://github.com/bmax121/APatch/issues/836)
> > [#544](https://github.com/bmax121/APatch/issues/544)
> 
> I am trying to rename the package and app name but it cannot open. Can you tell me any solution?

Until now, only solution is to hide the package manager via Hide My Applist (LSPosed module)

yoshimo, 2025-04-01:
Is that still necessary on modern android versions where applications can't easily query all installed packages?

VisionR1, 2025-04-01:
> Is that still necessary on modern android versions where applications can't easily query all installed packages?

Maybe in new Android have the permission QUERY_ALL_PACKAGES, and can close it and other apps can't see what install apps you have.

closed 2025-04-01 by ratankumardas