bmax121/APatch #1012: I installed shamiko in Apatch with a trick hahaha

Morrigan-Ship, 2025-04-04:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

in first root via
install zygisk-next and shamiko last version
then install Apatch
and flash boot made by Apatch
now goto modules section in Apatch and Enable shamiko :)

### Describe the solution you'd like/描述你想要的解决方案

.

### Describe alternatives you've considered/描述您考虑过的备选方案

.

VisionR1, 2025-04-04:
But working actually?
Because still can install it in APatch, if you know how.

VisionR1, 2025-04-04:
Understand, I'm asking, if after all this works.
Because can install in APatch if you modify and say work inside, but actually not working.

Morrigan-Ship, 2025-04-04:
it show `unsupported environment` 
i dont know it working well or not.
but i tested it in Kitsune and same error `unsupported environment`

i think if you update magisk api's it fix the problem
i can help you
The worst scenario is to edit the Shamiko Source to work, which I am currently helping

Morrigan-Ship, 2025-04-04:
can we contact in telegram?

VisionR1, 2025-04-04:
> it show `unsupported environment` i dont know it working well or not. but i tested it in Kitsune and same error `unsupported environment`
> 
> i think if you update magisk api's it fix the problem i can help you The worst scenario is to edit the Shamiko Source to work, which I am currently helping

Exactly.
Kitsune Mask is not supported from almost beginner.
I don't know how to this.
And shamiko is closed source.

Morrigan-Ship, 2025-04-04:
momo can find zygisk and magisk
it not work

VisionR1, 2025-04-04:
> momo can find zygisk and magisk it not work
> 
> EDIT: But we were able to get one step closer

Try and other new apps like native test, native root Detector, momo is old.
Many users who know at least the basic, have installed but not working, this must be fix it by devs. of shamiko.

Morrigan-Ship, 2025-04-04:
i tested
native test:

[image]

native root detector:

[image]

its full result i can see lsposed modules but cannot see magisk

Morrigan-Ship, 2025-04-05:
also `rootbeer sample` not found root or anything

VisionR1, 2025-04-05:
Do you have it in exclusion list? Or just shamiko install?

Morrigan-Ship, 2025-04-05:
i just installed shamiko

VisionR1, 2025-04-05:
Maybe work one part, service.sh.
Lot of users take only this.

Morrigan-Ship, 2025-04-05:
i checked service.sh it just change some values to hide everything
```
check_reset_prop "ro.boot.vbmeta.device_state" "locked"
check_reset_prop "ro.boot.verifiedbootstate" "green"
check_reset_prop "ro.boot.flash.locked" "1"
check_reset_prop "ro.boot.veritymode" "enforcing"
check_reset_prop "ro.boot.warranty_bit" "0"
...
```

if yes can we use this alone?

VisionR1, 2025-04-05:
> i checked service.sh it just change some values to hide everything
> 
> ```
> check_reset_prop "ro.boot.vbmeta.device_state" "locked"
> check_reset_prop "ro.boot.verifiedbootstate" "green"
> check_reset_prop "ro.boot.flash.locked" "1"
> check_reset_prop "ro.boot.veritymode" "enforcing"
> check_reset_prop "ro.boot.warranty_bit" "0"
> ...
> ```
> 
> if yes can we use this alone? it mean make a custom module? i try now!

Many users, already do that, have a folder service.d or boot.complete.d inside in the adb.

Morrigan-Ship, 2025-04-05:
> Many users, already do that, have a folder service.d or boot.complete.d inside in the adb

then what is the problem
we can use it instead of shamiko

VisionR1, 2025-04-05:
> > Many users, already do that, have a folder service.d or boot.complete.d inside in the adb
> 
> then what is the problem we can use it instead of shamiko

Is a simple service.d, just hide the very basic, the good hide is inside in the shamiko code.

Morrigan-Ship, 2025-04-06:
i find something :)
`post-fs-data.sh` this file:
it checking kernel su and magisk versions:
```
MAGISK_VERSION="$(magisk -V)"
KSU_VERSION="$(su -V)"
```
i tested in adb:
```
adb shell
su
su -V
11039

magisk -V
/system/bin/sh: magisk: inaccessible or not found
magisk binary no in $PATH so it cannot find it :/
so it error:
install new version of magisk 27005

Morrigan-Ship, 2025-04-06:
yessssssssss yes yes yesssss
i installed it without trick
hahaaaaaa

Morrigan-Ship, 2025-04-06:
i edited customize.sh and verify.sh and fixed

Morrigan-Ship, 2025-04-06:
ops
installed but after reboot it show unsupported environment ://

VisionR1, 2025-04-06:
Yeah i know this.
Users which installed, this way do it but after say this.

Morrigan-Ship, 2025-04-06:
i still working on it
i success to bypass the `Unsupported environment`
but i dont know how to sure it realy working or not!
i need help

Morrigan-Ship, 2025-04-06:
> How bypass

with hark work 😐😂
change all module.prop value
easy

VisionR1, 2025-04-06:
> > How bypass
> 
> with hark work 😐😂 change all module.prop value easy

That's is good. 👍🏻

Morrigan-Ship, 2025-04-06:
```
grep -nia "magisk" /proc/self/mountinfo
grep -nia "magisk" /proc/self/attr/prev
grep -nia "magisk" /proc/self/attr/current
```

some apps use these to find and detect zygisk, zygote, magisk
may help you

VisionR1, 2025-04-06:
> ```
> grep -nia "magisk" /proc/self/mountinfo
> grep -nia "magisk" /proc/self/attr/prev
> grep -nia "magisk" /proc/self/attr/current
> ```
> 
> some apps use these to find and detect zygisk, zygote, magisk may help you

Yeah.
Unfortunately lot of parameters can cause the detector, even with shamiko.

Morrigan-Ship, 2025-04-06:
but main problem with shamiko is that magisk -V: command not found because magisk is not in $PATH
make a new version with magisk in PATH i will test it

VisionR1, 2025-04-06:
> but main problem with shamiko is that magisk -V: command not found because magisk is not in $PATH make a new version with magisk in PATH i will test it

Understand.
Good luck with that.
In Kitsune Mask also can't install.

Morrigan-Ship, 2025-04-11:
well i found how it work
zygisk folder have prebuilt binaries for zygisk that are common files for all magisk modules
but i compared these files as binary

shamiko is patched `zygisk/arm64-v8a.so` and added a lib named `libshamiko.so`
if you open arm64-v8a.so via hex editor and search for `libshamiko.so` you will find it

VisionR1, 2025-04-11:
Yeah, i use HxD.

I think have some code inside in the zygisk, from main.cpp, if you compared with other open source projects.

closed 2025-05-14 by Admirepowered